chipsalliance/chisel #1211: Could you please write a tutorial on how to use Chisel3/firrtl in an offline PC?

hyf6661669, 2019-10-17:
For some reasons, I have to work on a computer without any internet-access. Therefore, I think the only way is downloading sbt/scala/chisel3/firrtl on my own computer and copy them to the offline computer. On my own computer, it seems that build.sbt will handle any ```libraryDependencies```. However, I don't know how to handle the dependecy problems without internet-access.

Much thanks!

schoeberl, 2019-10-17:
When you use a concrete version for the libraries you use, you only need Internet access the first time you build your project. At the second time the dependencies will be served by the cached files.

Cheers,
Martin

seldridge, 2019-10-17:
We probably need a guide on how to do this or some way of packaging things up... Currently, I'd try the following:
1. On the machine that has internet access, follow the instructions for the [Chisel Template](https://github.com/freechipsproject/chisel-template) getting through `sbt test`. After that, copy the contents of `~/.ivy2` from the internet access machine to the non-internet access machine. That should pre-populate the cache so everything will resolve locally. I have not tested this, however.
2. If you only need FIRRTL, you can run `sbt assembly` in a clone of the FIRRTL repository. This will populate `utils/bin/{firrtl, firrtl.jar}`. The former is a shell script calling the latter that you can use to generate Verilog from FIRRTL.

hyf6661669, 2019-10-17:
@seldridge 
If the OS of the non-internet access computer is windows, what should I do? Thanks!

hyf6661669, 2019-10-17:
It would be very nice if you can release a ZIP file for different OS. After downloading and unzipping this file, users can write chisel codes, do chisel simulation or generating verilog codes WITHOUT ANY INTERNET ACCESS.

edwardcwang, 2019-10-17:
From a development perspective: another way that we can realize this is to generate a Chisel fat JAR and then use that to compile Chisel code without Internet access.

colin4124, 2019-10-17:
https://github.com/colin4124/chisel3-releases/releases 

Have a try.  It is my way to use it  without Internet access.

seldridge, 2019-10-17:
@ucbjrl: Would it make sense for us to also package up a jar with each release like how @colin4124 is doing here?

ucbjrl, 2019-10-17:
I think that's a great idea.
I'll put it on my list.

closed 2019-10-31 by hyf6661669

joisty, 2019-12-24:
> https://github.com/colin4124/chisel3-releases/releases
> 
> Have a try. It is my way to use it without Internet access.

Would you please give a little more guide on how to use your jar files?
I have same problem, having to use chisel on an offline PC.
I used to do it  by copying .sbt & .ivy2 directory from an online PC.

colin4124, 2019-12-24:
@joisty 

I updated it just now, I try my best to explain it more details. If any step confuse you, feel free to point it out.

In short，there are two steps:

1. Get the all in one jar.
2. Put the jar to the according directory of your project.