chipsalliance/chisel #1212: Failed to bore submodule signal to current module

sequencer, 2019-10-21:
**Type of issue**: bug report

**Impact**: API addition (no impact on existing code) 

**Development Phase**: proposal

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
```scala
class Constant extends Module {
  val io = IO(new Bundle{})
  val x = Wire(UInt(6.W))
  x := 42.U
}
class Top extends Module {
  val io = IO(new Bundle{})
  val constant = Module(new Constant)
  val y = Wire(UInt(6.W))
  BoringUtils.bore(y, Seq(constant.x))
}
```

**What is the current behavior?**
```
  chisel3.internal.ChiselException: signalName/pathName should be called after circuit elaboration
    chisel3.internal.throwException$.apply(Error.scala:42)
    chisel3.internal.HasId.instanceName(Builder.scala:130)
    chisel3.internal.HasId.instanceName$(Builder.scala:124)
    chisel3.Data.instanceName(Data.scala:257)
    chisel3.util.experimental.BoringUtils$.genName$lzycompute$1(BoringUtils.scala:184)
    chisel3.util.experimental.BoringUtils$.genName$1(BoringUtils.scala:184)
    chisel3.util.experimental.BoringUtils$.$anonfun$bore$1(BoringUtils.scala:185)
    chisel3.util.experimental.BoringUtils$.$anonfun$bore$1$adapted(BoringUtils.scala:185)
    scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:238)
    scala.collection.immutable.List.foreach(List.scala:392)
    scala.collection.TraversableLike.map(TraversableLike.scala:238)
    scala.collection.TraversableLike.map$(TraversableLike.scala:231)
    scala.collection.immutable.List.map(List.scala:298)
    chisel3.util.experimental.BoringUtils$.bore(BoringUtils.scala:185)
```

**What is the expected behavior?**
Bore success.

**Please tell us about your environment:**
    - version: `3.2.0`

sequencer, 2019-10-21:
My fault, I miss placed the sink and source :(

seldridge, 2019-10-21:
Actually, @sequencer, while your example is backwards for source and sink, if you reverse the placement of these (put the source constant in the current module) this *is actually a bug*. There's a candidate fix in https://github.com/freechipsproject/chisel3/pull/1175.

sequencer, 2019-10-21:
thank you! @seldridge, I'm based on `chisel3 origin/bore-name`, `firrtl origin/v1.2.0` branch now.
It is quite strange that `WiringTransform` didn't run, while all `Annotation` exist, and I have no idea where this problem came from.

sequencer, 2019-10-21:
still failed with `firrtl: fd981848` and `chisel3: 22434659` :(

PR #1175 (2019-10-21): Internal BoringUtils.bore Bug Fix
This fixes a bug introduced in the original `BoringUtils.bore` implementation. This was using `instanceName` which meant that it could only be used for closed modules. The user-facing impact of this is an exception if you use `BoringUtils.bore` for signals in the module in which you do the bore (because that module is not closed). This adds one small test for this case.

This PR will introduce worse Verilog naming for boring signals using `BoringUtils.bore`.

Fixes #1212.

**Type of change**: bug report | other enhancement

**Impact**: no functional change

** [... cut]
closed 2019-10-21 by sequencer