chipsalliance/chisel #1230: Fill 0 returns type instead of value

jackkoenig, 2019-11-06:
**Type of issue**: bug report | feature request

**Impact**: no functional change (code could technically rely on this behavior but it would be super weird) 

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

`chisel3.util.Fill` when given `n = 0` returns `UInt(0.W)` which is a type rather than a value which it returns for `n >= 1`: https://github.com/freechipsproject/chisel3/blob/387274784115bee2bf7167547a7ee459219e1413/src/main/scala/chisel3/util/Bitwise.scala#L68

**What is the expected behavior?**

Ideally this would return a zero-width-wire which would then go away, but since support for that is lacking, it should probably just error in the meantime.

**What is the use case for changing the behavior?**

Better error messages and with zero-width wire support, cleaner generators that don't have to special case zero.

aswaterman, 2019-11-06:
I have been bitten by this.  Also, please fix it in such a way that `Cat(x, Fill(0, y), z)` is exactly equivalent to `Cat(x, z)`.  The broken status quo is actually better than it "working" but producing `Cat(x, 0.U, z)`.

jackkoenig, 2019-11-06:
Yeah, `Fill(0, y)` producing `0.U` is a big *no*, if it does not error, it must return `0.U(0.W)` which currently doesn't work but is the ideal result.