chipsalliance/chisel #1232: Bad Error Message for Connecting to Input

jackkoenig, 2019-11-06:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

```scala
import chisel3._

class Passthrough extends Module {
  val io = IO(new Bundle {
    val in = Input(UInt(8.W))
    val out = Output(UInt(8.W))
  })
  io.out := io.in
}

class MyModule extends Module {
  val io = IO(new Bundle {
    val in = Input(UInt(8.W))
    val out = Output(UInt(8.W))
  })
  val inst = Module(new Passthrough)
  inst.io.out := io.in
}

object MyTop extends App {
  chisel3.Driver.execute(args, () => new MyModule)
}
```

**What is the current behavior?**

You get the following generic error message:
```
[error] chisel3.internal.ChiselException: Connection between sink (UInt<8>(IO io_out in Passthrough)) and source (UInt<8>(IO in unelaborated MyModule)) failed @: Sink is unwriteable by current module.                                                                                                                   
[error]         ...
[error]         at MyModule.<init>(Fill.scala:18)
[error]         at MyTop$.$anonfun$new$1(Fill.scala:22)
```

**What is the expected behavior?**

We should be more specific in the error message

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

Better user experience with precise error messages

jackkoenig, 2019-11-06:
Note that this is even worse in aggregates because any leaf being the wrong direction, it just presents the generic error message about the full aggregate

mwachs5, 2020-01-31:
oooh. i ran into this.

jackkoenig, 2022-01-20:
This has been improve in Chisel 3.5.0!
```
chisel3.internal.ChiselException: Connection between sink (Passthrough.io.out: IO[UInt<8>]) and source (MyModule.io.in: IO[UInt<8>]) failed @: io.out in Passthrough cannot be written from module MyModule.
```
It still could be slightly more precise with a "because it is an input") but it's still pretty decent.

closed 2022-01-20 by jackkoenig