chipsalliance/chisel #1240: Different behavior between head and tail

starbrilliance, 2019-11-18:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
The function head and tail are often used to select sub-bits from a bit-vector. However, head(0) can emit correct verilog but tail(N) can NOT.

**What is the current behavior?**
For example, I write a test code like this:
```
class TestModule extends Module {
  val io = IO(new Bundle {
    val a = Input(UInt(8.W))
    val b = Output(UInt(8.W))
  })

  io.b := io.a.head(0)
}
```
And it will emit expected Verilog:
```
module TestModule(
  input        clock,
  input        reset,
  input  [7:0] io_a,
  output [7:0] io_b
);
  assign io_b = 8'h0; // @[test.scala 10:8]
endmodule
```
In contrast to it, tail(8) should emit same Verilog like this:
```
class TestModule extends Module {
  val io = IO(new Bundle {
    val a = Input(UInt(8.W))
    val b = Output(UInt(8.W))
  })

  io.b := io.a.tail(8)
}
```
But a bug occurs:
`
(run-main-0) firrtl.passes.CheckWidths$TailWidthException:  @[test.scala 9:20]: [target ~TestModule|TestModule] Parameter 8 in tail operator is larger than input width 8.
`
**What is the expected behavior?**
tail(N) should emit same code as head(0).
**Please tell us about your environment:**

- version: `3.2.0`
- OS: `Linux starbrilliance  5.0.0-32-generic #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`
**What is the use case for changing the behavior?**

sequencer, 2019-11-18:
This is a bug related to firrtl, since `fir` generated by chisel is correct.
I sent a PR there.

chick, 2019-12-19:
`tail` means something different than you think.
`tail` is equivalent in behavior to the `drop` method on a scala collection.
I.e. it drops the specified number of initial elements and returns the rest

closed 2019-12-19 by chick