chipsalliance/chisel #1287: Update logging to have name of design in "Elaborating design..."

oharboe, 2019-12-23:
**Type of issue**: feature request

**Impact**: no functional change

**Development Phase**: request | proposal

**What is the current behavior?**

Logging outputs a progress message without name of design beeing elaborated. 

```[info] [0.000] Elaborating design...```

**What is the expected behavior?**

Can we have the name of the design being elaborated, like:

```[info] [0.000] Elaborating ThingAMaBob...```

This can make tracing through logs a lot more pleasant. For instance, I discovered that chisel-testers2 does elaboration twice for Verilator, which would be a lot more evident if the logging specified which design is being elaborated.

colin4124, 2019-12-23:
The code relates to this issue as below:

https://github.com/freechipsproject/chisel3/blob/98a6710cc0447d79cbd12271ea450c70e619b6f8/chiselFrontend/src/main/scala/chisel3/internal/Builder.scala#L402-L410

The module you want to build is instantiated after output the info `Elaborating design...`. In other words, it can't know about any information of the module before it is instantiated.

So if you want to know about the name of design is elaborated, it can add the name to the done information.

```
[info] [0.000]  Done elaborating ThingAMaBob.
```

oharboe, 2019-12-23:
Much, much better, but not ideal. Elaboration can be nested. Ideally we'd want to know both at start and end which module is in equestion.

Nested elaborations can be useful. I use an elaboration and simulation to extract a property I'm using to set up a subsequent elaboration.

colin4124, 2019-12-23:
Do you mean？

```scala

class Foo extends Module {
   val m1 = Moudle(new Bar)
   val m2 = Module(new Car)
}

object Main extends App {
  Driver.execute(args, () => new Foo)
}
```

And then it prints:

```
[info] [0.000] Elaborating Foo...
[info] [0.000] Elaborating Bar...
[info] [0.000] Elaborating Car...
````

oharboe, 2019-12-23:
Not quite.

From inside elaboration of a module A, I run a chisel-tester2 test that extracts a property of the design of module B. This property can only be determined during simulation. This property (a size of a RAM) is then used to elaborate module A.

I realize this may be somewhat off the beaten path, but it is an example of where elaboration is nested.

colin4124, 2019-12-23:
I haven't used chisel-tester2,  I don't understand what you exactly want.

As I know,  take `val m0 = Module(new Mux2)` for example, if you want to get a property of the module `Mux2`, it must instantiated. 

https://github.com/freechipsproject/chisel3/blob/98a6710cc0447d79cbd12271ea450c70e619b6f8/chiselFrontend/src/main/scala/chisel3/Module.scala#L158-L168

You can a print statement after `Builder.currentModule = Some(this) ` which at the beginning of elaboration.

```scala
    Builder.currentModule = Some(this)
    Builder.errors.info(s"$name being created...")
    Builder.whenDepth = 0
```

The effect of this code as below:

```scala
import chisel3._

class Mux4 extends Module {
  ...
  val m0 = Module(new Mux2)
  ...
  val m1 = Module(new Mux2)
  ...
  val m3 = Module(new Mux2)
  ...
}
object Main extends App {
  Driver.execute(args, () => new Mux4)
}
```
```
[info] [0.000] Elaborating design...
[info] [0.002] Mux4 being created...
[info] [0.009] Mux2 being created...
[info] [0.017] Mux2_1 being created...
[info] [0.022] Mux2_2 being created...
[info] [0.030] Done elaborating Mux4.
```

oharboe, 2019-12-23:
What I want is the module name printed both on the "Elaborating ThingAMaBob" and "Done elaborating ThingAMaBob".

I run into nested elaboration when the property I need for module A can not be determined after elaboration of module B, *only after simulation* of module B, hence elaboration is nested.

colin4124, 2019-12-23:
Each time you call `Builder.build` method, it will print ""Elaborating design...", so maybe you can pass the design name as argument to `Builder.build` method and print it

oharboe, 2019-12-23:
I'm not invoking Builder.build(), I'm using chisel-testers2 to run a simulation, but I see what you mean.

Now, this feature request is about Chisel logging this information out of the box without any modification to application or Chisel code.