chipsalliance/chisel #1291: build.sc compilation fails using Chisel3 release-3.2.2 make with mill 0.5.4

aBaechtold, 2020-01-01:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Issue description**: 
Running `make` with `mill 0.5.4` (sbt not installed) will fail to compile/build Chisel3 release-3.2.2.
For details refer to the attached files (compilation failure related to `build.sc`).
Appears to be an API change on `mill`. Building Chisel3 with the default version that is fetched by the Makefile (mill version 0.3.5) works fine (despite of some warnings). 
Did not follow up in what version of mill this might have been introduced (suspect pull #634 or related).

**Other information**
To reproduce run:
`make MILL_BIN='_...your mill path..._' mill.publishLocal.all`

Attached archive contains ([IssueDetails.tar.gz](https://github.com/freechipsproject/chisel3/files/4014132/IssueDetails.tar.gz)):
- Bash CLI output running make with mill 0.5.4 (compilation errors)
- Bash CLI output running make with mill 0.3.5 (compilation is fine)
- Makefile and build.sc of Chisel3 release-3.2.2 for reference

**Environment:**
- OS: Manjaro Linux 18.1.4 XFCE, x86_64 
- Mill 0.5.4 (package details: Manjaro community package (non-AUR), built by Felix Yan on December/26/2019)

sequencer, 2020-01-01:
have a try on `mill-git`

aBaechtold, 2020-01-01:
Thanks for the reply. I think you are referring to the AUR package (of which you also seem to be the maintainer). I tried that package as you suggested. Using that version results in the same compilation failures; it behaves the same way as version 0.5.4.

Are you able to build Chisel3 with that package (i.e. there is something wrong with my setup, process) or does Chisel3 build.sc require an update to support the mill version in question?

Remark: The AUR package requires mill already installed (chicken-egg). In my case the installation failed at the installation step as the destination already existed.

sequencer, 2020-01-02:
I see.... I failed too...
It seems that `build.sc` of `chisel3` is out of date.
I'll send a PR here.

aBaechtold, 2020-02-26:
Hi all,
I fetched the latest code that should contain the fix and tried to build Chisel 3 (meanwhile Mill has been updated to v6.0.x, but I also tried with version 0.5.3). Unfortunately I get compilation errors. Am I doing something wrong ?

Build output (note that due to caching this does not contain every entry of an initial mill build run): 
[BuildOutput.txt](https://github.com/freechipsproject/chisel3/files/4258233/BuildOutput.txt)

Git status:
```
[abaechto@DellXPS chisel3]$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   .gitignore

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	.metals/
	BuildOutput.txt

no changes added to commit (use "git add" and/or "git commit -a")
[abaechto@DellXPS chisel3]$ 
```

sequencer, 2020-02-26:
You didn't have `firrtl` in your `localIvy`.
If you wanna use `chisel3` and `firrtl` for project only(without `publishLocal`)
You can try
```
object myfirrtl extends firrtl.build.firrtlCrossModule("2.12.10") {
  override def millSourcePath = super.millSourcePath / 'firrtl
}

object mychisel3 extends chisel3.build.chisel3CrossModule("2.12.10") {
  override def millSourcePath = super.millSourcePath / 'chisel3
  def firrtlModule: Option[PublishModule] = Some(myfirrtl)
}
```
for example https://github.com/sequencer/rocket-playground/blob/master/build.sc

If you wanna publish a specific version of `chisel3` to `localIvy`, you should publish `firrtl` firstly:
```
mill firrtl[2.12.10].publishLocal
``` 
Then publish chisel3
```
mill chisel3[2.12.10].publishLocal
```

ucbjrl, 2020-02-26:
We're way behind publishing chisel3 and firrtl SNAPSHOTs, but in any case,
as suggested in a previous email. if you're trying to build chisel3 from
master source, you should first build and publishLocal firrtl from master
source.

edwardcwang, 2020-02-27:
Also, make sure you are using at least mill 0.6.0. The latest build.sc PRs by Sequencer work well for me.

closed 2022-03-23 by jackkoenig