chipsalliance/chisel #1292: Chisel creates unused _RAND_* signals which verilator complains about

antonblanchard, 2020-01-03:
I'd like to build my project with `verilator -Wall`, but chisel creates `_RAND_*` regs. A simple example:

```
import chisel3._

class Foo extends Module {
  val io = IO(new Bundle {
    val a = Input(UInt(1.W))
    val b = Output(UInt(1.W))
  })

  val r = RegInit(0.U(1.W))
  r := io.a
  io.b := r
}

object FooObj extends App {
  chisel3.Driver.execute(Array[String](), () => new Foo)
}
```

Results in:

```
module Foo(
  input   clock,
  input   reset,
  input   io_a,
  output  io_b
);
  reg  r; // @[Foo.scala 9:18]
  reg [31:0] _RAND_0;
  assign io_b = r; // @[Foo.scala 11:8]
`ifdef RANDOMIZE_GARBAGE_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_INVALID_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_REG_INIT
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_MEM_INIT
`define RANDOMIZE
`endif
`ifndef RANDOM
`define RANDOM $random
`endif
`ifdef RANDOMIZE_MEM_INIT
  integer initvar;
`endif
`ifndef SYNTHESIS
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
  `ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  r = _RAND_0[0:0];
  `endif // RANDOMIZE_REG_INIT
  `endif // RANDOMIZE
end // initial
`endif // SYNTHESIS
  always @(posedge clock) begin
    if (reset) begin
      r <= 1'h0;
    end else begin
      r <= io_a;
    end
  end
endmodule
```

Using an empty verilator driver:

```
int main(int argc, char **argv)
{
}
```

and building with:

```
verilator -O3 -Wall --assert --cc Foo.v --exe Foo.cpp
```

we hit the following error:

```
%Warning-UNUSED: Foo.v:8: Signal is not driven, nor used: _RAND_0
                 ... Use "/* verilator lint_off UNUSED */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
%Error: Command Failed /usr/bin/verilator_bin -O3 -Wall --assert --cc Foo.v --exe Foo.cpp
```

We are already wrapping everything in `ifdef RANDOMIZE_MEM_INIT` etc, so we should just wrap the reg with an ifdef too.

seldridge, 2020-01-07:
I agree with your analysis and proposed solution to move the register definition inside the `ifdef`. :+1: Thanks for digging into it. What your proposing here would be a pretty straightforward FIRRTL PR modifying the `VerilogEmitter`. I'm fine to keep this open for tracking purposes here.

seldridge, 2020-05-14:
Fixed via https://github.com/freechipsproject/firrtl/pull/1548.

This will be available in Chisel 3.3.1 (FIRRTL 1.3.1) which should be out today or tomorrow.

closed 2020-05-14 by seldridge