chipsalliance/chisel #1295: Incorrectly sized enum results in confusing and verbose error message

antonblanchard, 2020-01-05:
I have noticed some common mistakes result in huge backtraces. One example is an incorrectly sized Enum:

```
import chisel3._
import chisel3.util._

class Bar extends Module {
  val io = IO(new Bundle {
  })

  val a :: b :: c :: d :: Nil = Enum(3)
}

object BarObj extends App {
  chisel3.Driver.execute(Array[String](), () => new Bar)
}
```

Gives the following error message:

```
[info] [0.002] Elaborating design...
[error] (run-main-0) firrtl.options.OptionsException: Exception thrown when elaborating ChiselGeneratorAnnotation
[error] firrtl.options.OptionsException: Exception thrown when elaborating ChiselGeneratorAnnotation
[error]         at chisel3.stage.ChiselGeneratorAnnotation.elaborate(ChiselAnnotations.scala:55)
[error]         at chisel3.stage.phases.Elaborate.$anonfun$transform$1(Elaborate.scala:19)
[error]         at scala.collection.TraversableLike.$anonfun$flatMap$1(TraversableLike.scala:245)
[error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at scala.collection.TraversableLike.flatMap(TraversableLike.scala:245)
[error] oo / Comat scala.collection.TraversableLike.flatMap$(TraversableLike.scala:242)
[error] 	at scala.collection.immutable.List.flatMap(List.scala:355)
[error] 	at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:18)
[error] 	at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:16)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:32)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Translator.transform(Phase.scala:140)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:140)
[error] 	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error] 	at chisel3.stage.ChiselStage.$anonfun$run$2(ChiselStage.scala:33)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at chisel3.stage.ChiselStage.run(ChiselStage.scala:33)
[error] 	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error] 	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:32)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Translator.transform(Phase.scala:140)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:140)
[error] 	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:46)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:46)
[error] 	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:168)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at logger.Logger$.makeScope(Logger.scala:166)
[error] 	at firrtl.options.Stage.transform(Stage.scala:46)
[error] 	at firrtl.options.Stage.transform(Stage.scala:17)
[error] 	at chisel3.Driver$.$anonfun$execute$1(Driver.scala:226)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at chisel3.Driver$.execute(Driver.scala:226)
[error] 	at chisel3.Driver$.execute(Driver.scala:249)
[error] 	at BarObj$.delayedEndpoint$BarObj$1(Bar.scala:12)
[error] 	at BarObj$delayedInit$body.apply(Bar.scala:11)
[error] 	at scala.Function0.apply$mcV$sp(Function0.scala:39)
[error] 	at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[... 60 lines cut ...]
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at chisel3.Driver$.execute(Driver.scala:226)
[error] 	at chisel3.Driver$.execute(Driver.scala:249)
[error] 	at BarObj$.delayedEndpoint$BarObj$1(Bar.scala:12)
[error] 	at BarObj$delayedInit$body.apply(Bar.scala:11)
[error] 	at scala.Function0.apply$mcV$sp(Function0.scala:39)
[error] 	at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[error] 	at scala.App.$anonfun$main$1$adapted(App.scala:80)
[error] 	at scala.collection.immutable.List.foreach(List.scala:392)
[error] 	at scala.App.main(App.scala:80)
[error] 	at scala.App.main$(App.scala:78)
[error] 	at BarObj$.main(Bar.scala:11)
[error] 	at BarObj.main(Bar.scala)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
```

The answer to the problem is there, but it's hiding 54 deep in the backtrace:

```Bar.scala:11```

It would be also nice to have a more specific error, eg `Enum is too small to fit values defined`

seldridge, 2020-01-07:
Unfortunately, this is really a Scala error that I think is hard to tease out. 

Under the hood, `Enum(n)` is just a method that generates a `List[UInt]`. The pattern shown of doing this assignment via a List unapply, while terse, is perhaps overly confusing. And, due to the fact that this is relying on something built into Scala it's, likely hard to check.

Consider the following (which is basically the same as above):

```scala
val a :: b :: Nil = 1 :: Nil // Or: val List(a, b) = List(1)
```

This is doing some pattern magic (which I wasn't able to find a good reference for on how this actually works) in order to do the assignment to the vals. This would be roughly equivalent to the following: 

```scala
val (a, b) = List(1) match { 
  case _a :: _b :: Nil => (_a, _b)
}
```

Running this on the ammonite REPL (which is going to provide nice truncation) you wind an error like you see in Chisel. However, here the error isn't surprising (I tried to pattern match a two-element list with an incomplete set of patterns that could only match a one-element list):

```
scala.MatchError: List(1) (of class scala.collection.immutable.$colon$colon)
  ammonite.$sess.cmd0$.<clinit>(cmd0.sc:1)
```

So, with all that waffling and lack of a good answer... I'd suggest you see if `EnumFactory` is more to your liking. There is no great documentation of this, but you should be able to follow along with the examples in the tests: https://github.com/freechipsproject/chisel3/blob/4ad624265e2b8aef5771af71006431705bdb19df/src/test/scala/chiselTests/StrongEnum.scala

antonblanchard, 2020-01-08:
Thanks @seldridge! The lack of a "proper" enum did surprise me when I first started playing with Chisel, so I should definitely take a look. Assuming that is the way forward, I'm happy to close this out.