chipsalliance/chisel #1297: asUInt for bundle literals may be inconsistent

zhuanhao-wu, 2020-01-10:
**Type of issue**: bug report | other enhancement

**Impact**: API modification

**Development Phase**: request

**Other information**
 
I would expect `asUInt` to give results that honor the width of the `Bundle` so that I can use it to convert a bundle literal to UInt consistently.

That said, in the following test won't pass. `bundleLitUInt` should have the same `litValue` as `anotherBundleLitUInt` because the two bundle literals both have `a=3, b=3`.
```scala
  class MyMultipleWidthBundle extends Bundle {
    val a = UInt(3.W)
    val b = UInt(2.W)
  }
  "bundle literals asUInt" should "work" in {
    assertTesterPasses(new BasicTester {
      val wire = Wire(UInt(5.W))
      val anotherWire = Wire(UInt(5.W))
      val bundleLitUInt = new MyMultipleWidthBundle().Lit(
        _.a -> 3.U(2.W), _.b -> 3.U(3.W)
      ).asUInt
      val anotherBundleLitUInt = new MyMultipleWidthBundle().Lit(
        _.a -> 3.U, _.b -> 3.U
      ).asUInt

      wire := bundleLitUInt
      anotherWire := anotherBundleLitUInt
      chisel3.assert(wire === anotherWire)
      stop()
    })
```

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
Consider `MyMultipleWidthBundle` in the above example: 
- Bundle literals created as `anotherBundleLitUInt`, in which we don't specify the widths of `UInt` in `Lit()`, will not honor the field widths of the bundle. The resultant UInt in the above example is thus 15 (`11 11` in binary).
- Bundle literals created as `bundleLitUInt` will honor the widths. The resultant UInt in the above example is thus 27 (`11 011` in binary).

**What is the expected behavior?**
Both `anotherBundleLitUInt` and `bundleLitUInt` should be 27.

**Please tell us about your environment:**
- version: `3.3-SNAPSHOT`, `3.2-SNAPSHOT`
- OS: `Darwin sy 18.7.0 Darwin Kernel Version 18.7.0: Thu Jun 20 18:42:21 PDT 2019; root:xnu-4903.270.47~4/RELEASE_X86_64 x86_64`

**What is the use case for changing the behavior?**
This can be helpful when I use bundle literals as the input to a Verilog module. This way I don't need to use a wrapper to slice the ports.

ducky64, 2020-01-10:
In your example, the Bundle definition has `a` of width 3 and `b` of width 2, but the bundle literal has this reversed (`a` of width 2, `b` of width 3). Is this intentional or a mistake? I'd expect the correct answer to be `011 11` = 15, but it's hard to differentiate that from `11 11` = 15.

In any case, there's at least one bug in there (that it's possible to specify a fixed width literal that ignores the Bundle's structural width).

zhuanhao-wu, 2020-01-10:
@ducky64 Sorry and yes, there's a typo in my example: the width of `a` should be 2 and that of `b` should be 3. I fixed it and updated the description. Still, the code will not pass from my side.