chipsalliance/chisel #1298: Nested when/.elsewhen statement hits Scala warning

antonblanchard, 2020-01-10:
I've turned some Scala warnings on, and nested `when/.elsewhen` statements are showing warnings. An example:

In build.sbt:

```scalacOptions ++= Seq("-Ywarn-value-discard")```

And this test case:

```
import chisel3._

class Test extends Module {
  val io = IO(new Bundle {
    val a = Input(Bool())
    val b = Input(Bool())
    val c = Input(Bool())
    val d = Output(Bool())
  })

  when (io.a) {
    when (io.b) {
      io.d := true.B
    } .elsewhen (io.c) {
      io.d := false.B
    }
  }
}

object TestObj extends App {
  chisel3.Driver.execute(Array[String](), () => new Test)
}
```

I see:

```
[warn] main/scala/Test.scala:14:24: discarded non-Unit value
[warn]     } .elsewhen (io.c) {
[warn]                        ^
```

colin4124, 2020-01-11:
I try to explain that.

Here are simplest code to reproduce this issue.

```scala
class WhenNestCase extends Module {
  val io = IO(new Bundle {
    val a = Input(Bool())
    val b = Input(Bool())
    val c = Input(Bool())
    val d = Output(Bool())
  })

  io.d := false.B
  when (io.a) {
    when (io.b) {
      io.d := true.B
    }
  }
}

object Main extends App {
  Driver.execute(args, () => new WhenNestCase)
}
```

`when` is a **Object**:

```scala
object when {
  def apply(cond: => Bool)(block: => Unit): WhenContext = {
    new WhenContext(Some(() => cond), block)
  }
}
```

Let's de-sugar:

```scala
when(io.a)({
    when (io.b)({
      io.d := true.B
    })
  })
```

The first `when` object take two arguments: `io.a` and other `when` object. 

**`when` object return value `WhenContext`**, so the `when(io.b) {...}` has non-Unit value

seldridge, 2020-01-11:
It's probably reasonable to change when to `apply[A](cond: => Bool)(block: => A): WhenContext` without any effects.

closed 2020-01-22 by mergify[bot]

antonblanchard, 2020-01-22:
Thanks @seldridge!

PR #1308 (2020-01-22): Change when/switch thunk type to Any
This changes the type of the thunk used by `when` and `switch` method from `=> Unit` to `=> Any`.

This avoids a warning related to discarding a non-Unit value (`-Ywarn-value-discard` ) if you do something like what is describe in #1298.

Alternatively, and as I suggested in #1298, I think you could also get rid of this by making the methods polymorphic in the type of the thunk, e.g., `apply[A](a: => A): FooContext`. However, this is not like `withClock` where the type of the thunk is returned. Therefore, I opted for the simpler solution of using `Any`.

** [... cut]