chipsalliance/chisel #1307: Problems with Interval Inference

chick, 2020-01-21:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**What is the current behavior?**
The following chisel test
```scala
  class IntervalInferenceTester extends BasicTester {
    val start = WireInit(1.0.I(range"[-16.0,15.0].5"))
    val blankWire = Wire(Interval())
    blankWire := start
    chisel3.assert(blankWire === start)
    stop()
  }
```
with the error.
```
 @[IntervalSpec.scala 965:15]: [module IntervalInferenceTester]  Type mismatch in 'blankWire <= start'.
  blankWire: Interval.0
  start: Interval[-16, 15].5
firrtl.passes.CheckTypes$InvalidConnect:  @[IntervalSpec.scala 965:15]: [module IntervalInferenceTester]  Type mismatch in 'blankWire <= start'.
```

**What is the expected behavior?**
Seems like inference should manage to match the type of a first order connection

chick, 2020-01-21:
@azidar, Seems like this should pass to me. I'm a bit worried that tests of this sort will fail
for a lot of inference scenarios. Or am I missing something

azidar, 2020-01-22:
Does `Interval()` generate firrtl that has zero precision? If so, it should generate firrtl with unknown precision.

chick, 2020-01-22:
I have a PR not yet submitted that does just that.