chipsalliance/chisel #1314: Mux1H generates functionally different RTL depending on data type

johningalls-sifive, 2020-01-27:
**Type of issue**: bug report
**Impact**: API modification
**Development Phase**: request

**What is the current behavior?**
this Chisel that looks formally equivalent:
```
val sel = Wire(Vec(1, Bool()))
val in = Wire(Vec(1, Bool()))
val out1 = Mux1H(sel, in)
val out2 = Mux1H(sel.asUInt, in,asUInt)
```
produces different output RTL that is **not** formally equivalent:
```
assign out1 = in;
assign out2 = sel & in;
```
`out1` is generated by https://github.com/freechipsproject/chisel3/blob/387274784115bee2bf7167547a7ee459219e1413/chiselFrontend/src/main/scala/chisel3/SeqUtils.scala#L53
`out2` is generated by https://github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/util/Mux.scala#L30

**What is the expected behavior?**
**What is the use case for changing the behavior?**
`out1` and `out2` should generate the same output RTL.  I care more that it's formally equivalent than which logic gets generated.

aswaterman, 2020-01-27:
Looks like the second case should return in if either in’s or sel’s width is KnownWidth(1).

johningalls-sifive, 2020-02-06:
I'd almost prefer that Mux1H generate the same logic regardless of width, i.e. don't special-case the 1-wide case in SeqUtils.scala, so that the And gate is still instantiated in the AndOrMux.
This would effectively define the behavior of Mux1H to be an AndOrMux, where the output is zero if the selects are all zero.  I've been guilty of making this assumption for logic optimization.

aswaterman, 2020-02-06:
I definitely don’t want to change the behavior of the 1-wide case for the
other ones.

If you want the and-or thing even in the 1-wide case, it’s a one-liner in
Scala, or you can make your own library method for it rather than using the
Chisel one.

johningalls-sifive, 2020-03-02:
> I definitely don’t want to change the behavior of the 1-wide case for the other ones. If you want the and-or thing even in the 1-wide case, it’s a one-liner in Scala, or you can make your own library method for it rather than using the Chisel one.

will do

closed 2020-03-02 by johningalls-sifive