chipsalliance/chisel #1317: java.io.IOException: Cannot run program "dot": error=2, No such file or directory

luoxinjie1146, 2020-02-02:
0_demo.ipynb
// same 3-point moving average filter as before
visualize(() => new FirFilter(8, Seq(1.U, 1.U, 1.U)))
[info] [0.000] Elaborating design...
[info] [0.071] Done elaborating.
Total FIRRTL Compile Time: 159.2 ms

java.io.IOException: Cannot run program "dot": error=2, No such file or directory
java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.scala:75)
scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuilderImpl.scala:104)
scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuilderImpl.scala:103)
dotvisualizer.FirrtlDiagrammer$.render(FirrtlDiagrammer.scala:93)
dotvisualizer.transforms.ModuleLevelDiagrammer.execute(ModuleLevelDiagrammer.scala:184)
dotvisualizer.transforms.MakeDiagramGroup.execute(MakeDiagramGroup.scala:48)
ammonite.$file.dummy.source.load$minusivy_2$Helper.generateVisualizations(Main.sc:155)
ammonite.$file.dummy.source.load$minusivy_2$Helper.visualize(Main.sc:173)
ammonite.$sess.cmd5$Helper.(cmd5.sc:1)
ammonite.$sess.cmd5$.(cmd5.sc:7)
ammonite.$sess.cmd5$.(cmd5.sc:-1)
java.io.IOException: error=2, No such file or directory
java.lang.UNIXProcess.forkAndExec(Native Method)
java.lang.UNIXProcess.(UNIXProcess.java:247)
java.lang.ProcessImpl.start(ProcessImpl.java:134)
java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.scala:75)
scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuilderImpl.scala:104)
scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuilderImpl.scala:103)
dotvisualizer.FirrtlDiagrammer$.render(FirrtlDiagrammer.scala:93)
dotvisualizer.transforms.ModuleLevelDiagrammer.execute(ModuleLevelDiagrammer.scala:184)
dotvisualizer.transforms.MakeDiagramGroup.execute(MakeDiagramGroup.scala:48)
ammonite.$file.dummy.source.load$minusivy_2$Helper.generateVisualizations(Main.sc:155)
ammonite.$file.dummy.source.load$minusivy_2$Helper.visualize(Main.sc:173)
ammonite.$sess.cmd5$Helper.(cmd5.sc:1)
ammonite.$sess.cmd5$.(cmd5.sc:7)
ammonite.$sess.cmd5$.(cmd5.sc:-1)

chick, 2020-02-03:
This requires installation of GraphViz. Information is available here. [https://graphviz.gitlab.io/download/](https://graphviz.gitlab.io/download/)

closed 2020-02-21 by jackkoenig

Burnleydev1, 2021-11-07:
Thank you @chick, I had the very problem and this solved it.