chipsalliance/chisel #1350: Naming of ready/valid interface

schoeberl, 2020-02-19:
**Type of issue**:  feature request 

**Impact**:  API addition (no impact on existing code), later API modification 

I would like to give the ready/valid interface a better name and also a better nome for the data field.

Interface name: It is not clear to me what DecoupledIO means. Who is decoupling from whom? A more logical name for the ready/valid interface would be ReadyValidIO.

Data field: this field is normally (e.g., in Dally) simply named data. Why bits?

I know renaming might be painful, but it has been done. And if it gives in the long run better reading (code understanding) it might be worth it.

albert-magyar, 2020-02-19:
In Chisel, there exist both `ReadyValidIO` and `DecoupledIO` classes. The fact that both classes already exist (and that there is semantic meaning to the distinction) would be an extreme roadblock to renaming.

Here, `DecoupledIO <: ReadyValidIO`. The intent is that `ReadyValidIO` means "a bundle that has ready and valid signals, along with bits." This could be associated with a number of different handshake timings that exist.

`DecoupledIO` additionally adds the constraint that the interface obeys a classic "fire on ready and valid" constraint. This is why it adds a concrete implementation of `fire` that is purely combinational logic.

There are other subclasses of `ReadyValidIO`, including `IrrevocableIO`, which enforces FIFO ordering. One can imagine other flavors, such as one that implicitly includes the guarantee that `ready` not depend on `valid` for sinks (as required for AXI).

The idea behind the particular naming scheme is that `ReadyValidIO` is a good candidate for the most general form of the bundle, since anything with a ready / valid interface could easily be called a `ReadyValidIO`.

As for the choice of `bits`, there's no arguing that it's a bit quirky. It would be possible to "alias" it with a `def`, but that might do more harm than good in obscuring how it works.

albert-magyar, 2020-02-19:
There is [a blurb in the wiki](https://github.com/freechipsproject/chisel3/wiki/frequently-asked-questions#why-decoupledio-instead-of-readyvalidio) to this effect, but it admittedly is a bit weak.

schoeberl, 2020-02-19:
> On 19 Feb, 2020, at 22:23, Albert Magyar <notifications@github.com> wrote:
> 
> In Chisel, there exist both ReadyValidIO and DecoupledIO classes. The fact that both classes already exist (and that there is semantic meaning to the distinction) would be an extreme roadblock to renaming.
> 

Oh, I overlooked that one. Yes, this is a roadblock. So we can close this issue ;-)
> Here, DecoupledIO <: ReadyValidIO. The intent is that ReadyValidIO means "a bundle that has ready and valid signals, along with bits." This could be associated with a number of different handshake timings that exist.
> 
Mmh, I think it is not good to allow “a number of different handshake timings”. ready/valid has a meaning that when both are asserted the transfer takes place. It would be super irritating using these names and do something completely different, like (a made up example): I assert valid only once, and you tell me later with ready that you can take another one.

> There are other subclasses of ReadyValidIO, including IrrevocableIO, which enforces FIFO ordering. One can imagine other flavors, such as one that implicitly includes the guarantee that ready not depend on valid for sinks (as required for AXI).
> 
> 
As far as I understand, the irrevocable interface marks that the producer will keep data and valid set until the consumer asserts a ready.
> The idea behind the particular naming scheme is that ReadyValidIO is a good candidate for the most general form of the bundle, since anything with a ready / valid interface could easily be called a [...]
Ok, we have to live with this name. I think.

Cheers,
Martin

schoeberl, 2020-02-20:
I change my mind: that DecoupledIO extends ReadyValidIO is *not* a roadblock, its a bonus. If we can simply drop the abstract from ReadyValidIO, it would become usable. (And this would make me happy ;-)

jackkoenig, 2020-02-21:
We don't have to make ReadyValidIO non abstract (although we could), we could just make it have a companion object that returns a concrete subtype (like DecoupledIO)

schoeberl, 2020-02-22:
That’s an option as well. Maybe a matter of taste. I am more used to create objects with new instead of having a companion object with an apply method. Simply less code for (just) IO definitions. But maybe this is just me.

Cheers,
Martin

jackkoenig, 2020-02-22:
While in most circumstances it's a matter of taste, there are advantages to the factory method pattern:
1. You can refactor the factory method to change the concrete subtype
2. You can refactor the factory method to do setup and teardown before and after object creation

For code maintainability reasons I think we should use factory methods and even make constructors private where possible.

schoeberl, 2020-02-22:
Ok, convinced. For important interfaces/classes (or library code in general) it makes sense to do this little extra step to provide a factory method.