chipsalliance/chisel #1351: Chisel3 (local) compile error

schoeberl, 2020-02-20:
**Type of issue**: bug report

Compiling Chisel 3 locally gives me a compile error:

```
/Users/martin/data/repository/chisel/chisel3/src/main/scala/chisel3/Driver.scala:8:8: object Dependency is not a member of package firrtl.options
[error] import firrtl.options.{Dependency, Phase, PhaseManager, StageError}
[error]        ^
[error] /Users/martin/data/repository/chisel/chisel3/src/main/scala/chisel3/stage/phases/DriverCompatibility.scala:7:8: object Dependency is not a member of package firrtl.options
[error] import firrtl.options.{Dependency, OptionsException, OutputAnnotationFileAnnotation, Phase, PreservesAll, Unserializable}
```
Do I need also to compile publish locally FIRRTL to compile chisel3? I did this and compiled without an issue. But ```sbt compile``` in chisel3 still fails.

**Please tell us about your environment:**

macOS Catalina, JDK 8

seldridge, 2020-02-20:
You probably need to bump your local FIRRTL repo and publish that. There was an API breaking change introduced in https://github.com/freechipsproject/firrtl/pull/1357 with Chisel updated/fixed in https://github.com/freechipsproject/chisel3/pull/1270.

If you're working off of the master branch then that may require working with the master branches of subproject dependencies. The master branch includes API breakages targeted at the next release, here 3.3. if you want the most up-to-date bug fixes you can work off the 3.2.x branch which is supposed to be API and binary compatible with 1.2.+ FIRRTL.

schoeberl, 2020-02-20:
> You probably need to bump your local FIRRTL repo and publish that. There was an API breaking change introduced in freechipsproject/firrtl#1357 <https://github.com/freechipsproject/firrtl/pull/1357> with Chisel updated/fixed in #1270 <https://github.com/freechipsproject/chisel3/pull/1270>.

I did clone firrtl and did: sbt compile, sbt publishLocal in master. Shouldn't Chisel master working with firrtl master?

seldridge, 2020-02-20:
> I did clone firrtl and did: sbt compile, sbt publishLocal in master.

Try doing an `sbt clean` inside `chisel3` and recompiling. Failing that, try blowing away `~/.ivy2/cache` and `~/.ivy2/local`. Then re-publishing FIRRTL. [SBT has weird resolution semantics](https://github.com/sbt/sbt/issues/2687).

I can reproduce the behavior you see if I:

```bash
rm -rf ~/.ivy2/cache ~/.ivy2/local
git clone git@github.com:freechipsproject/chisel3
(cd chisel3 && sbt compile) # <- This errors because it uses remote SNAPSHOT

git clone git@github.com:freechipsproject/firrtl
(cd firrtl && sbt publishLocal)
(cd chisel3 && sbt compile) # <- This still errors

(cd chisel3 && sbt clean && sbt compile) # <- This works
```

For the record, I'm using the following versions:

  - FIRRTL: 6766c035f3b77d4793ef28041f08588c9faa5473 (Wed Feb 19 20:55:27 2020 -0500)
  - Chisel3: 403cd5b950e8865b41279cadffb640e7cb2d651c (Wed Feb 19 21:23:36 2020 -0500)

> Shouldn't Chisel master working with firrtl master?

CI checks this before any Chisel PR is merged. Ergo, this should *always* be true. If you see something like this, there's a high probability that it's some issue locally.

schoeberl, 2020-02-20:
> Chisel3: 403cd5b <https://github.com/freechipsproject/chisel3/commit/403cd5b950e8865b41279cadffb640e7cb2d651c> (Wed Feb 19 21:23:36 2020 -0500)

Thanks for your help and explanation! I dumped my whole ~./.ivy2 and sbt clean in chisel3 did the trick.

Cheers,
Martin

seldridge, 2020-02-20:
Nice. Glad it's working for you.

closed 2020-02-20 by seldridge