chipsalliance/chisel #1352: MultiIO modules which have foo := reset get an IO called foo instead of reset

mwachs5, 2020-02-20:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Example: (https://scastie.scala-lang.org/cgfAvZWGSqCGa5D309sKuA)

```
import chisel3._

class MyModule extends MultiIOModule {
  val in = IO(Input(UInt(8.W)))
  val out = IO(Output(UInt(8.W)))

  val other = reset
  out := in

object Main extends App {
  println(chisel3.Driver.emitVerilog(new MyModule))
}

}
```
produces:

```
module MyModule(
input        clock,
input        other,
input  [7:0] in,
output [7:0] out
);
assign out = in; // @[main.scala 9:7]
endmodule
```

**What is the current behavior?**

If you assign something (e.g. `foo`) to `reset` inside your module, the I/O name changes from `reset` to `foo`

**What is the expected behavior?**

The I/O name stays as `reset` and there is just an assignment and use to an internal value called `foo`.

**Please tell us about your environment:**

Chisel hash: `* 5ee1efa - (24 hours ago) Restore SNAPSHOT version. (#1349) - Jim Lawson (HEAD, origin/3.2.x)`

**What is the use case for changing the behavior?**

Stability and predictability of I/O names

mwachs5, 2020-02-20:
note a workaround is making the `val foo` a `private val foo`

jackkoenig, 2020-09-30:
This is fixed in v3.4.0-RC3 https://scastie.scala-lang.org/qRlQUSKLRfiPasiCseDyYQ

closed 2020-09-30 by jackkoenig