chipsalliance/chisel #1354: VecInit doesn't work with 0-length vector

mwachs5, 2020-02-21:
**Type of issue**: bug report | feature request

**Impact**: API modification | unknown

**Development Phase**: request 

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

This works:

```scala
import chisel3._

val vec_length = 0
 val reg_pmp = RegInit({
    
    val resetValue = WireInit(Vec(vec_length, Bool()), DontCare)
    resetValue.foreach{ _ := false.B}
    resetValue
  })
```

This does not (throws a require exception):

```scala
val vec_length = 0
 val reg_of_vec = RegInit(VecInit(Seq.fill(vec_length) {false.B}))
```

**What is the current behavior?**

Trying to initialize a zero-length vector with VecInit throws an exception, while doing so with Vec(0,...) does work

**What is the expected behavior?**

VecInit(Seq.fill(0){...}) with zero-length vector should work as well as Vec (0,...)

**Please tell us about your environment:**

Chisel hash:
`* | | 060b421 - (4 weeks ago) Fixed code example typo in comment (#1294) (#1313) - mergify[bot] (HEAD)`

*What is the use case for changing the behavior?**

Being able to follow the cookbook pattern of :

`val reg = RegInit(VecInit(Seq...))` to initialize vectors, regardless of their size.

edwardcwang, 2020-02-22:
There may be a semantic problem: what should the following code (which is equivalent to `Seq.fill(0){x}` mean?

```scala
val questionable_literal = VecInit(Seq())
```

aswaterman, 2020-02-22:
@edwardcwang It should (and will) fail because the Scala compiler won’t infer a concrete type for the Seq’s type parameter. So is there actually a problem?

edwardcwang, 2020-02-22:
If there is consensus that there is not actually a problem, then I believe that we can resolve this issue.

mwachs5, 2020-02-23:
> There may be a semantic problem: what should the following code (which is equivalent to Seq.fill(0){x} mean?
> 
> val questionable_literal = VecInit(Seq())
> Should it infer Vec(0, Bool()) appropriately "initialized"?

No, I don't think `VecInit(Seq())` should infer `Vec(0, Bool())`. 
But I don't agree that `Seq.fill(0){x}` assuming you are saying `x = Bool()`, is equivalent to that.

What if I did:
```
val empty_bool_seq: Seq[Bool] = Seq()
VecInit(empty_bool_seq)
```
Isn't that different than `VecInit(Seq())`?

> Should it be able to be connected to a Vec(0, UInt(16.W))?
> etc.

No, it shouldn't. But it should be allowed to be connected to a Vec(0, UInt(16.W)) or more importantly I should be able to do something like:

`questionable_literal.foreach { _ := false.B }`

edwardcwang, 2020-02-24:
Hmm, as I understand it, Andrew's point is that `VecInit(Seq())` should fail since it appears to be ill-defined as no type can be inferred for the Vec.

One issue is that with Scala's type erasure, can we actually detect something like `VecInit(Seq.empty[Bool])`? And how do we specify parameters like the width of an `UInt` since I don't believe it's possible to write something like `VecInit(Seq.empty[UInt(16.W)])`?

mwachs5, 2020-02-24:
I am missing something in what you are saying @edwardcwang . Let's say it's true what you are saying that VecInit(Seq()) is ill defined. But then why is it fine to do what *does* work with `WireInint(Vec(0,...)`? And if the pattern endorsed by the cookbook doesn't generally work, we should be cookbooking a different pattern ( see the `val initRegOfVec` example at https://github.com/freechipsproject/chisel3/wiki/Cookbook#how-do-i-create-a-reg-of-type-vec)

mwachs5, 2020-02-24:
Also, I am not sure it matters "how can we specify parameters like the width of an UInt since I don't believe it's possible to write something like VecInit(Seq.empty[UInt(16.W)])?"

If I understand chisel correctly, the type is just UInt. So, it should correctly let you write code like:

```my_empty_uint_vec = WireInit(VecInit(Seq.fill(0){UInt(16.W)}))
my_other_empty_uint_vec = WireInit(VecInit(Seq.fill(0){UInt(4.W)}))
```
But in what case would you ever actually assign something to it so that the width matters?

```
my_empty_uint_vec := my_other_empty_uint_vec // type check is fine, doesn't matter since nothing ever actually gets hooked up
```

edwardcwang, 2020-02-24:
The issue seems to be that `Seq.fill(0){UInt(4.W)}` is the same as `Seq.empty[UInt]` which is different from `Vec(0, UInt(4.W))` where the type is explicitly specified.

edwardcwang, 2020-02-24:
Resolution from dev meeting today:

* Support `VecInit(Seq.empty[T]) => Vec[T]`
  * But not for any indexing (including dynamic)
  * Must throw exception to avoid `sample_element` problem
  * Generates no hardware
* Stretch goal, support Nothing so that VecInit(Seq.empty[MyBundle])(0.U(0.W)) is legal

aswaterman, 2020-02-24:
This doesn't contradict what you wrote, but connections to/from and iteration (foreach etc.) should be legal (and generate no hardware).

ducky64, 2020-02-24:
Yeah, both those cases were discussed and should be support-able. I think fundamentally, since a zero length Vec is known as such during elaboration time, it can runtime error on any attempt to materialize an element. Iteration (including map, foreach, and in a connect) on an empty Vec would gate attempting to materialize an element and should work fine.

There's a few interpretations of when Vec errors. One is that static indexing is bounds-checked but dynamic indexing is not (currently the case in chisel; but dynamic index on a zero-length Vec will fail in FIRRTL). Another is that any indexing is checked to the maximum degree possible - so while we can't check indexing a (dynamic, Wire) UInt a non-zero-length Vec (especially if the Vec is non-power-of-two length), we know what any access to a zero-length Vec would fail.

Food for thought though - with Intervals, we have tighter bounds on index values, so theoretically it would be possible to have stricter dynamic bounds checks.

jackkoenig, 2023-05-09:
`VecInit.fill` now supports `0`, `VecInit(Seq())` is not possible with how Chisel works.

closed 2023-05-09 by jackkoenig