chipsalliance/chisel #1362: Current Implementation of Counter makes it impossible to create a 32bit or larger counter

claford-v-lawrence, 2020-02-26:
**Type of issue**: feature request 

**Impact**: API modification

**Development Phase**:  proposal

**Other information**
 
I submitted a PR

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
val x = Module(new Counter(4294967296L))
val x = Module(new Counter(4294967296))
these doesn't work

**What is the current behavior?**
Counter does not work with integers other than Int

**What is the expected behavior?**
Counter should work with long etc

**Please tell us about your environment:**

**What is the use case for changing the behavior?**
enabling all Counters to be able to handle large ints and longs without breaking backward compatibility

aswaterman, 2020-02-26:
This seems desirable, but would be nice to figure out how to do it without breaking API compatibility (w.r.t. `Counter.n`).

albert-magyar, 2020-02-26:
I imagine the easiest non-API breaking way to do it would be with a rename of the actual internal BigInt counter. A type parameter and the `Numeric` type class would be cool, but that is going to end up getting exposed some way or another.

Hilariously, this may be binary-compatible.
```scala
 class Counter(val max: BigInt) {
   def this(intMax: Int) {
     this(BigInt(intMax)) // or just rely on implicits
   }

   def n: Int = {
     val truncated = max.toInt
     val msg = "Viewing Counter.n as Int caused an overflow. Use Counter.max!"
     assert(BigInt(truncated) == max, msg)
     truncated
   }
  ...
  // change all remaining refs to 'n' to use 'max' in inc
```

albert-magyar, 2020-02-26:
@claford-v-lawrence, I think it would be good to re-open your PR. Maybe you could incorporate some adapters to smooth the API change?

claford-v-lawrence, 2020-02-26:
Sure. I'll keep digging if there is a better way to handle the changes.

albert-magyar, 2020-02-26:
Great! If it's decided that it's necessary to keep around `n` as an `Int` in the short term, I think something more complete version of the sketch above could work, which refactors the counter around using a `BigInt` but preserves the API by keeping `n` around as a `def` of type `Int`.

This pushes some errors (calling `.n` on a counter that would not fit in an `Int`) to runtime, though.

claford-v-lawrence, 2020-02-27:
Thanks for the help. I'm using some other work around to avoid this problem :)