chipsalliance/chisel #1364: Manipulating Bundles in tests

phhorrein, 2020-02-27:
**Type of issue**: feature request

**Impact**: unknown

**Development Phase**: request

**Other information**
 

I have run in an issue when trying to match data representation in testers and modules under tests. Here is what I have done:

- I have defined a Bundle, which represents an header of some kind, which I use in my module
- The interface to my module is a serialized interface, which manipulates UInt. 

What I want to do know is to generate some meaningful headers in my tests, and then send them to the module. I have tried the following solutions:

- create a BundleLiterals, and then get the litOption/litValue -> not implemented
- create a BundleLiterals, and then use the asUInt macro, and use range selection in the UInt -> not working as it requires a Builder context

The only thing I can do for now is to generate a BigInt manually from the tests. 

So the question is, how can I do this properly?

- version: `3.2.3`
- OS: `Linux laptop1143701 4.15.0-88-generic #88-Ubuntu SMP Tue Feb 11 20:11:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

edwardcwang, 2020-02-27:
What goes wrong with using the current Chisel bundle literals support?

phhorrein, 2020-02-27:
Sorry, my explanation was not that clear: I'm not sending the litteral directly, I'm serializing it, sending it chunk by chunk over a FIFO interface. This means that the interface is not aware of typing, it's only an UInt. Typing is only known inside the test, and inside the module.