chipsalliance/chisel #1366: SyncReadMem.read(x) and Mem.read(RegNext(x)) behave differently

davidmetz, 2020-03-02:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
[TagSetTest.txt](https://github.com/freechipsproject/chisel3/files/4274643/TagSetTest.txt)
This module and test implement a tag set (essentially just the tag portion of a 2-way cache).
The provided test fails when using SyncReadMem.
When using Mem or Reg combined with registers for either the address or data it works.
This behaviour occurs for both treadle and verilator.

**What is the current behavior?**
SyncReadMem.read(x) and Mem.read(RegNext(x)) behave differently.
In the example provided tag_tables(0) is never read when using SyncReadMem.
**What is the expected behavior?**
That they behave the same
**Please tell us about your environment:**

I'm using chipyard release 1.1.0 on Linux Mint.
**What is the use case for changing the behavior?**
Code working as expected

edwardcwang, 2020-03-02:
Is your goal to explicitly construct a synchronous-read memory?

albert-magyar, 2020-03-02:
Currently, `SyncReadMem.read` has complicated enable semantics that produce wildly odd behavior for some corner cases. There have been some "spot patches" proposed for cases like yours (see #840), but the plan of record is to resolve this with an API change for 1.4.0, so this should be appearing in master shortly after the 1.3.0 release.

The canonical way to work around this is to make sure the signal you're using as the address for a `SyncReadMem` read is assigned to a wire.

Here, a simple concrete solution would be:
```
val idx = WireDefault(index(io.s0_check(i).bits))
```
This forces `idx` to be an assignable wire instead of a continuous expression "node".

closed 2020-03-02 by albert-magyar