chipsalliance/chisel #1376: Functional mismatch between chisel and its generated verilog

rayjyh, 2020-03-12:
I converted the following chisel

>  12 class GCD extends Module {
>  13   val io = IO(new Bundle {
>  14     val a        = Input(UInt(16.W))
>  15     val b        = Input(UInt(16.W))
>  16     val sel1     = Input(UInt(2.W))
>  17     val out      = Output(UInt(16.W))
>  18   })
>  19 
>  20   when(io.sel1 === 0.U) { io.out := io.a }
>  21   when(io.sel1 === 1.U) { io.out := io.b }
>  22   .elsewhen(io.sel1 === 2.U) {io.out := io.a     + 1.U}
>  23   .otherwise {io.out := io.b + 1.U}
>  24 }
> 

to verilog and got 

>   1 module GCD(
>   2   input         clock,
>   3   input         reset,
>   4   input  [15:0] io_a,
>   5   input  [15:0] io_b,
>   6   input  [1:0]  io_sel1,
>   7   output [15:0] io_out
>   8 );
>   9   wire  _T_1; // @[GCD.scala 26:16]
>  10   wire  _T_2; // @[GCD.scala 27:21]
>  11   wire [15:0] _T_4; // @[GCD.scala 27:46]
>  12   wire [15:0] _T_6; // @[GCD.scala 28:30]
>  13   wire [15:0] _GEN_0; // @[GCD.scala 27:30]
>  14   assign _T_1 = io_sel1 == 2'h1; // @[GCD.sca    la 26:16]
>  15   assign _T_2 = io_sel1 == 2'h2; // @[GCD.sca    la 27:21]
>  16   assign _T_4 = io_a + 16'h1; // @[GCD.scala     27:46]
>  17   assign _T_6 = io_b + 16'h1; // @[GCD.scala     28:30]
>  18   assign _GEN_0 = _T_2 ? _T_4 : _T_6; // @[GC    D.scala 27:30]
>  19   assign io_out = _T_1 ? io_b : _GEN_0; // @[    GCD.scala 25:34 GCD.scala 26:34 GCD.scala 27:    38 GCD.scala 28:22]
>  20 endmodule
> 
In verilog, when `io.sel` is 0,` io.out` is `io.b+1`, which doesn't make sense. I think the compiler should ban the same condition signal to be controlled by multiple `when`s.

schoeberl, 2020-03-12:
This is OK and consistent with other languages like VHDL or Verilog. Last assignment counts.

Cheers,
Martin

jackkoenig, 2020-03-16:
Closing as this is known and documented (albeit perhaps not as prominently as it should be) behavior: https://www.chisel-lang.org/chisel3/combinational-circuits.html#wires

closed 2020-03-16 by jackkoenig