chipsalliance/chisel #1386: Request for standard operator for "assign superclass to subclass" or "assign by name"

mwachs5, 2020-03-24:
**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: request 

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Not a bug

**What is the current behavior?**

This example doesn't compile with  `chisel3` when does for `Chisel._`:

```
trait HasALotOfComplicatedSignals {.... many vals here}
class BundleWithALotOfComplicatedSignals extends Bundle with HasALotOfComplicatedSignals
class BundleWithEvenMoreSignals extends HasALotOfComplicatedSignnals {
  val foo = Bool()
}
val subclassBundle = WireInit(new BundleWithEvenMoreSignals().fromBits(0.U))
val superclassBundle = WireInit(new BundleWithALotOfComplicatedSignals().fromBits(0.U))
subclassBundle := superclassBundle // This doesn't work in chisel3
subclassBundle.foo := false.B
```

**What is the expected behavior?**

There is some standard operator or method for doing :=_by_name which just takes all the signals from the right that match the ones on the left and assign them, leaving the others unconnected.

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

Making it possible and standard to bulk assign from superclass to a subclass

jackkoenig, 2020-03-24:
There's a very old issue for this: https://github.com/freechipsproject/chisel3/issues/661

The problem is that doing this in a semi-safe way (ie. not just reverting to unsafe Chisel 2 semantics) is tricky to implement.

mwachs5, 2020-03-24:
I guess. I suppose if there was a specific operator that meant "I am knowingly assigning from superclass to subclass" it might be less unsafe? ANyway I'll read the discussion on #661

mwachs5, 2021-08-27:
I think this is closed by #1955

closed 2021-08-27 by mwachs5