chipsalliance/chisel #1398: Ergonomics of asUInt

jackkoenig, 2020-04-03:
`asUInt` is defined as a 0-arity method on `Data` with parentheses. This means it can be called with, or without parentheses:
```scala
x.asUInt
x.asUInt()
```

This is fine and dandy, but it does mean that you can't call it without parentheses and then immediately call `apply` to do bit extraction on the resulting `UInt`:

```scala
x.asUInt(4) // no arguments allowed for nullary method chars: ()UInt
```
This error is obvious to more experienced users, but not to newer users and is just generally subpar ergonomics.

My proposal is to migrate to parentheses-less `asUInt`. I worried this couldn't be done but the wizards of StackOverflow showed me that it can: https://stackoverflow.com/questions/61004457

We can define a version of `asUInt` that takes varargs and deprecate it, then redefine the 0-arity one with no parentheses. This might be complicated by the macros so that's the big caveat to this idea.

**Type of issue**: other enhancement

**Impact**: API modification

**Development Phase**: proposal

azidar, 2020-10-27:
Linter solution for 3.4.x?

jackkoenig, 2021-10-13:
Coincidentally, this is resolved by https://github.com/chipsalliance/chisel3/pull/2124

closed 2021-10-13 by jackkoenig