chipsalliance/chisel #1401: How about providing `pwhen` or parallelWhen

davidmlw, 2020-04-07:
https://github.com/freechipsproject/chisel3/blob/dbb024a9adee6d82f37e357cf8b55456674ff65c/core/src/main/scala/chisel3/When.scala#L12

There is an popular design pattern in Chisel like this, and the statements can be mixed in from anywhere. in most cases in this design pattern, the when condition are parallel, that is, the order of mixing-in doesn't matter. While the current `when` semantic is a serial implementation resulting a priority Mux chain, which will introduce redundent logics.
`
data := initial_data
when(condA) { data := dataA }
when(condB) { data := dataB }
when(condC) { data := dataC }
`

How about providing `pwhen` or parallelWhen, where the conditions in `pwhen` are parallel and the implementation will be Mux1H. A `pwhen` has no `otherwise` or `elsewhen` statements. 
`
data := initial_data
pwhen(condA) { data := dataA }
pwhen(condB) { data := dataB }
pwhen(condC) { data := dataC }
`

davidmlw, 2020-04-07:
And it is better to have an option in Chisel that in some mode, `pwhen` equals to `when` for some purpose.

johningalls-sifive, 2020-04-07:
Yes, this design pattern does come up often.
Another way to do this could be to specify a `default` for `Mux1H` when zero selects are hot.

davidmlw, 2020-04-07:
I just realize it is not easy to implement pwhen considering when and pwhen are nested each other. A better way would be ultize DontCare.
`data := initial_data`
`when(condA) { data := dataA } .otherwise { data := DontCare }`
`when(condB) { data := dataB } .otherwise { data := DontCare }`
`when(condC) { data := dataC } .otherwise { data := DontCare }`

johningalls-sifive, 2020-04-07:
> Yes, this design pattern does come up often.
> Another way to do this could be to specify a `default` for `Mux1H` when zero selects are hot.

specifcally...
```
object Mux1HDefault {
  def apply[T <: Data](in: Iterable[(Bool, T)], default: T): T = {
    Mux1H(in ++ Seq(!in.map(_._1).reduce(_ || _) -> default))
  }
}
```

davidmlw, 2020-04-08:
> I just realize it is not easy to implement pwhen considering when and pwhen are nested each other. A better way would be ultize DontCare.
> `data := initial_data`
> `when(condA) { data := dataA } .otherwise { data := DontCare }`
> `when(condB) { data := dataB } .otherwise { data := DontCare }`
> `when(condC) { data := dataC } .otherwise { data := DontCare }`
> This should be equal to pwhen above.
> 
> So the question is how to optimize the assignments tree with DontCares that `when` statement collects. Or has the current DontCare supported this?

This would be a major modification of `DontCare` or `when` sementic, and not straightforward. One problem is that, when more than one condition is true, how `data` is assigned. In this case, it is unclear.

So this design pattern may have to introduce `pwhen`, with such features:
1. pwhen can be degraded to when by chisel option.
2. pwhen emits dynamic assertions to monitor the conditions are one-hot.
3. pwhen has to be related the assignment target, either designed in API or automatically collected and analyzed by Chisel/Firrtl.
4. pwhen and when are nested each other: the rule "once pwhen, always pwhen" perhaps works.