chipsalliance/chisel #1403: chiselName behaving badly when nesting modules

phhorrein, 2020-04-07:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

When using chiselName on two modules, one of which is nested in the other, chiselName is only applied on outermost module. This leads to strange generated Verilog. 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

The following code shows the issue:
```scala
class innerInnerModule extends Module {
  val io = IO(new Bundle {
    val in_1 = Input(UInt(5.W))
    val in_2 = Input(UInt(5.W))
    val out_1 = Output(UInt(5.W))
  })
  
  val outReg = Reg(chiselTypeOf(io.out_1))
  outReg := io.in_1 + io.in_2
  io.out_1 := outReg
}   
   
@chiselName 
class innerModule extends RawModule {
  val clock = IO(Input(Clock()))
  val reset = IO(Input(Bool()))
  val in_1 = IO(Input(UInt(5.W)))
  val in_2 = IO(Input(UInt(5.W))) 
  val out_1 = IO(Output(UInt(5.W))) 
  
  withClockAndReset(clock, reset) {
    val innermost = Module(new innerInnerModule())
    innermost.io.in_1 := in_1
    innermost.io.in_2 := in_2
    out_1 := innermost.io.out_1
  }
} 

@chiselName
class outerModule extends RawModule {
  val clock = IO(Input(Clock()))
  val reset = IO(Input(Bool()))
  val in_1 = IO(Input(UInt(5.W)))
  val in_2 = IO(Input(UInt(5.W)))
  val out_1 = IO(Output(UInt(5.W)))
  
  withClockAndReset(clock, reset) {
    val duplicated = Module(new innerInnerModule())
    duplicated.io.in_1 := in_1
    duplicated.io.in_2 := in_2
    out_1 := duplicated.io.out_1
  } 
  
  val inner = Module(new innerModule())
  inner.clock := clock
  inner.reset := reset
  inner.in_1 := in_1
  inner.in_2 := in_2
  out_1 := inner.out_1
} 
  
object outerMain extends App{
  (new chisel3.stage.ChiselStage).execute(
    Array("-X", "verilog"),
    Seq(ChiselGeneratorAnnotation(() => new outerModule())))
} 
```
**What is the current behavior?**

The generated verilog for innerModule is the following: 

```verilog
module innerModule(
  input        clock,
  input  [4:0] in_1,
  input  [4:0] in_2,
  output [4:0] out_1
);  
  wire  innerInnerModule_clock; // @[chiselNameIssue.scala 27:27]
  wire [4:0] innerInnerModule_io_in_1; // @[chiselNameIssue.scala 27:27]
  wire [4:0] innerInnerModule_io_in_2; // @[chiselNameIssue.scala 27:27]
  wire [4:0] innerInnerModule_io_out_1; // @[chiselNameIssue.scala 27:27]
  innerInnerModule innerInnerModule ( // @[chiselNameIssue.scala 27:27]
    .clock(innerInnerModule_clock),
    .io_in_1(innerInnerModule_io_in_1),
    .io_in_2(innerInnerModule_io_in_2),
    .io_out_1(innerInnerModule_io_out_1)
  );
  assign out_1 = innerInnerModule_io_out_1; // @[chiselNameIssue.scala 30:11]
  assign innerInnerModule_clock = clock;
  assign innerInnerModule_io_in_1 = in_1; // @[chiselNameIssue.scala 28:23]
  assign innerInnerModule_io_in_2 = in_2; // @[chiselNameIssue.scala 29:23]
endmodule
```

**What is the expected behavior?**

I would expect the generated verilog to look like this, as generated when using innerModule instead of outerModule in the App. 
```verilog
module innerModule(
  input        clock,
  input        reset,
  input  [4:0] in_1,
  input  [4:0] in_2,
  output [4:0] out_1
);
  wire  innermost_clock; // @[chiselNameIssue.scala 28:27]
  wire [4:0] innermost_io_in_1; // @[chiselNameIssue.scala 28:27]
  wire [4:0] innermost_io_in_2; // @[chiselNameIssue.scala 28:27]
  wire [4:0] innermost_io_out_1; // @[chiselNameIssue.scala 28:27]
  innerInnerModule innermost ( // @[chiselNameIssue.scala 28:27]
    .clock(innermost_clock),
    .io_in_1(innermost_io_in_1),
    .io_in_2(innermost_io_in_2),
    .io_out_1(innermost_io_out_1)
  );
  assign out_1 = innermost_io_out_1; // @[chiselNameIssue.scala 31:11]
  assign innermost_clock = clock;
  assign innermost_io_in_1 = in_1; // @[chiselNameIssue.scala 29:23]
  assign innermost_io_in_2 = in_2; // @[chiselNameIssue.scala 30:23]
endmodule
```

**Please tell us about your environment:**
    - version: `3.3-SNAPSHOT`
    - OS: `Linux baldr 4.15.0-91-generic #92-Ubuntu SMP Fri Feb 28 11:09:48 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

-->
**What is the use case for changing the behavior?**

jackkoenig, 2020-04-07:
This exists in `v3.3.0-RC1`

I believe this is a real regression introduced by #1209 (as confirmed by reverting it on master and regaining the `v3.2.4` behavior). I'm not certain of the precise cause, but I believe it's with the behavior of naming the module instance in:
```scala
  withClockAndReset(clock, reset) {
    val innermost = Module(new innerInnerModule())
    innermost.io.in_1 := in_1
    innermost.io.in_2 := in_2
    out_1 := innermost.io.out_1
  }
```
The `val` name that `v3.2.4` gives is `innermost` which is what is expected. Chisel has long had a default name for instances which is just the name of the Module (ie. `innerInnerModule` which is what we get in `v3.3.0-RC1`).

jackkoenig, 2020-04-07:
@ducky64 is this something you could look at?

johnsbrew, 2020-04-08:
Hi @jackkoenig 
Sorry for introducing so many side effects with the very small change of #1209 
I guess it is due to the return of current `this`in:

https://github.com/freechipsproject/chisel3/pull/1209/files#diff-48971ed501d2b5bb351ceb65542f71c8R131

I might have a look, at least add @phhorrein testcase in the specs and see if I really understand what's going on.

jackkoenig, 2020-04-10:
@johnsbrew no worries, all code changes carry some risk. I think this is a good thing to fix soon if we can, but we do what we can.

jackkoenig, 2020-12-02:
I can confirm that while this bug still exists in v3.3.2 (https://scastie.scala-lang.org/pwkaC42iSFOH1LbUM3gnBw) it is fixed by using the compiler plugin in v3.4.0 (https://scastie.scala-lang.org/zlBUdgS8R9WukJshx8KDHQ)

closed 2020-12-02 by jackkoenig