chipsalliance/chisel #1409: Error when reference defined in when scope is accessed outside of it

jackkoenig, 2020-04-10:
If you use a `var` to have a node escape it's when context (see https://github.com/freechipsproject/chisel3/pull/1408), you can refer to it and generate invalid FIRRTL. Now FIRRTL incorrectly accepts it (see https://github.com/freechipsproject/firrtl/issues/1505), but both chisel3 and FIRRTL should error on this.

**Type of issue**: feature request

**Impact**: API modification(?) bug bugfix kind

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

See #1408 

**What is the current behavior?**

chisel3 emits invalid FIRRTL

**What is the expected behavior?**

chisel3 should error if the user writes code that is invalid FIRRTL

**What is the use case for changing the behavior?**

Correctness

albert-magyar, 2020-05-04:
Also see freechipsproject/firrtl#1528 for the fix to the scoping in FIRRTL.

jackkoenig, 2020-09-30:
Fixed in v3.4.0-RC3: https://scastie.scala-lang.org/DT9TlX5ISKaR2pYZSLQQ4Q

closed 2020-09-30 by jackkoenig