chipsalliance/chisel #1410: Private vals don't get named by reflection nor @chiselName

jackkoenig, 2020-04-10:
The most egregious example is in `Queue` which uses private vals. Now perhaps we should just make them public, but the issue still stands.

Long ago, Chisel reflective naming worked on `private val` (https://github.com/freechipsproject/chisel3/pull/407), it seems this is no longer true.

It's also of note that `@chiselName` doesn't name private vals.

I feel like both of these are bugs that should be fixed.

**Type of issue**: bug report

**Impact**:  API addition (no impact on existing code)

**Development Phase**: request

mwachs5, 2020-04-10:
There was a workaround for some other issue that was relying on this behavior.... let me find which one it is

seldridge, 2020-08-03:
(Having not tested it and just looking through issues/PRs...) How does the Chisel naming plugin handle this? Does that close this issue?

jackkoenig, 2020-08-05:
Just checked, the plugin does indeed handle this!

Fixed by https://github.com/freechipsproject/chisel3/pull/1448

closed 2020-08-05 by jackkoenig