chipsalliance/chisel #1419: Blackbox instance input port will set to zero instead of report a uninitialized error if it isn't be assigned

s1104439112, 2020-04-25:
**Type of issue**: bug report | feature request | documentation | other enhancement
bug report

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal
request
**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
`
// Test.scala
// ultraRAM BlackBox
class URAM(aw: Int, dw: Int) extends BlackBox(Map("aw" -> aw, "dw" -> dw))
  with HasBlackBoxResource {
  val io = IO(new Bundle {
    val clk = Input(Clock())
    val we = Input(Bool())
    val waddr = Input(UInt(aw.W))
    val din = Input(UInt(dw.W))
    val raddr = Input(UInt(aw.W))
    val dout = Output(UInt(dw.W))
  })
  addResource("/ultraRAM.v")
}
// test module
class Test extends Module{
  val io = IO(new Bundle{
    val in = Input(Bool())
    val out = Output(UInt(8.W))
  })
  val temp = Module(new URAM(8, 8))
  io.out := 7.U
}
// output Verilog
object GetVeilogTest extends App {
  chisel3.Driver.execute(Array("--target-dir", "test_run_dir/make_test_verilog"), () => new Test)
}
`
**What is the current behavior?**
In above Test module circuit, I don't assign URAM BlackBox IO, but output verilog show that URAM's input is set to zero. In a word, a blackbox instance input port will set to zero if it isn't be assigned.

**What is the expected behavior?**
Expected behavior is a error will be proposed by firrtl like uninitialized.
 A regular module written by Chisel3 instead of BlackBox will be detected uninitialized error.

**Please tell us about your environment:**

- version:`3.2.5`
-OS:`Linux archlinux 5.6.6-arch1-1 #1 SMP PREEMPT Tue, 21 Apr 2020 10:35:16 +0000 x86_64 GNU/Linux`

**What is the use case for changing the behavior?**
I think blackbox input port need be treaded as wire and if a wire is not be assigned, firrtl will report this error. So, does it is a special design for solving some problems?
Thanks in advance!
[image]
[image]

sequencer, 2020-04-25:
Would you mind providing `ultraRAM.v` for reproducing?

s1104439112, 2020-04-25:
Sorry, I forgot it. I have appended ultraRAM.v in reproduce. It is a stand simple-dual RAM with vivado attribute.

sequencer, 2020-04-25:
This is my behavior:
```scala
import $ivy.`edu.berkeley.cs::chisel3:latest.integration`
import chisel3._
import chisel3.experimental._
import chisel3.util._

// ultraRAM BlackBox
class URAM(aw: Int, dw: Int) extends BlackBox(Map("aw" -> aw, "dw" -> dw))
  with HasBlackBoxInline {
  val io = IO(new Bundle {
    val clk = Input(Clock())
    val we = Input(Bool())
    val waddr = Input(UInt(aw.W))
    val din = Input(UInt(dw.W))
    val raddr = Input(UInt(aw.W))
    val dout = Output(UInt(dw.W))
  })
  setInline("BlackBoxRealAdd.v",
    s"""module URAM(
       |  clk,
       |  raddr,
       |  din,
       |  waddr,
       |  dout,
       |  we
       |);
       |
       |parameter aw = 16;
       |parameter dw = 16;
       |
       |input clk;
       |input we;
       |input [aw-1: 0] raddr;
       |input [aw-1: 0] waddr;
       |input [dw-1: 0] din;
       |output reg [dw-1: 0] dout;
       |
       |(* ram_style="ultra" *)
       |reg [dw-1: 0] mem[(1 << aw) - 1 : 0];
       |
       |always@(posedge clk)
       |begin
       |  if(we)
       |    mem[waddr] <= din;
       |    dout <= mem[raddr];
       |end
       |
       |endmodule""".stripMargin)
}
// test module
class Test extends MultiIOModule{
  val in = IO(Input(Bool()))
  val out = IO(Output(UInt(8.W)))
  val temp = Module(new URAM(8, 8))
  out := 7.U
}

(new chisel3.stage.ChiselStage).emitVerilog(new Test())
```
and generate such verilog:
```
module Test(
  input        clock,
  input        reset,
  input        in,
  output [7:0] out
);
  URAM #(.aw(8), .dw(8)) temp ( // @[PlayGround.sc 53:20]
  );
  assign out = 8'h7; // @[PlayGround.sc 54:7]
endmodule
```
Seems we should guard Verilog generation more strictly on the chisel side.

albert-magyar, 2020-04-25:
Summary:
* This is a known behavior and, for better or for worse, is part of the API
* Unrelated: avoid wildcard `import chisel3.experimental._` at all costs

While I do agree that this is inconsistent, this is actually intentional and not a bug per se.

The discrepancy is actually unrelated to Verilog generation -- it is due to the way Chisel generates different FIRRTL for blackbox instantiations which invalidates all IO even in the presence of `import chisel3.ExplicitCompileOptions.Strict` in the parent context.

Consider the following DFF model definitions:
```scala
import chisel3._
import chisel3.util._
import chisel3.ExplicitCompileOptions.Strict

class DFFVerilog extends BlackBox with HasBlackBoxInline {
  val io = IO(new Bundle {
    val clk = Input(Clock())
    val d = Input(Bool())
    val q = Output(Bool())
  })
  setInline("DFF.v", "???")
}

class DFFNative extends Module {
  val io = IO(new Bundle {
    val clk = Input(Clock())
    val d = Input(Bool())
    val q = Output(Bool())
  })
  val r = RegNext(io.d)
  io.q := r
}
```

If you instantiate `DFFNative` and don't explicitly connect to any of its IO, you will get the following FIRRTL:
```
    inst dff of DFFNative
    dff.clock <= clock
```
This will error in FIRRTL due to lack of initialization if it isn't caught in Chisel first (which is sometimes, but not always the case).

If you instantiate `DFFVerilog` and don't explicitly connect to any of its IO, you will get the following FIRRTL:
```
    inst dff of DFFVerilog
    dff.q is invalid
    dff.d is invalid
    dff.clk is invalid
```

In other words, Chisel explicitly chooses to drive all inputs with the abstract, undefined default value of "Invalid" through the `IsInvalid` statement ([implemented here](https://github.com/freechipsproject/chisel3/blob/9f620e06bacc2882068adfd4972ec2e9a87ea723/core/src/main/scala/chisel3/BlackBox.scala#L90) in `BlackBox.scala`). This has the effect of making the compilation error-free, **even though I have imported `chisel3.ExplicitCompileOptions.Strict`**. This may not be the best choice, but it is an explicitly supported facet of the API, and removing it would break people's code. The original justification is that blackboxes are themselves free of the burden of driving all of their outputs, so connections to blackboxes already follow a different mental model with respect to initialization.

This may be a good potential future change to the behavior of `Strict`. I will leave that to later discussion, but it would need to be a carefully managed API change.

The handling of `IsInvalid` and the default coercion to zero is a a separate concern -- while there may be better defaults, this can be overridden by adding a different default in a transform, and may be addressed in the future. It also is not specific to blackboxes.

s1104439112, 2020-04-26:
Thanks sequencer and albert-magyar spend time on my problem. I am cleanr now.

closed 2020-04-26 by s1104439112