chipsalliance/chisel #1422: Implicit Int2Bool conversion returns Bool(false) for all non null int values expect 1

johnsbrew, 2020-04-28:
**Type of issue**: bug report 
**Impact**:  API modification 
**Development Phase**: request

**Other information**

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

```scala
import chisel3._
```

**What is the current behavior?**
```scala
johnsbrew@ 65.B
res18: Bool = Bool(false)
```

**What is the expected behavior?**
```scala
johnsbrew@ 65.B
res18: Bool = Bool(true)
```

**Environment:**
    - version: `3.3.0-RC3`
    - OS: `Darwin 17.7.0 Darwin Kernel Version 17.7.0: Tue Feb 18 22:51:29 PST 2020; root:xnu-4570.71.73~1/RELEASE_X86_64 x86_64`

**What is the use case for changing the behavior?**
correctness: implicit conversion should be:
```scala
implicit def int2Bool(i: Int) : Bool = Bool(i != 0)
```

```scala
when(128728.B) {
    printf("should be displayed")
}
```

albert-magyar, 2020-05-03:
[This is currently an elaboration-time error.](https://scastie.scala-lang.org/gnT06SJ2SByJVkS2BFYj7w) The return value of `Bool.Lit(false)` only serve to avoid throwing an exception, which nominally lets elaboration continue before reporting the fatal error.

Admittedly, there isn't a great dividing line between what kinds of errors are accumulated this way vs. immediately thrown; this is mostly a legacy code issue. These "accumulated errors" rely on a hardware elaboration context to be reported, which explains what you see in the REPL. Making it possible to write Chisel code in an arbitrary context is a worthy goal, but there are unfortunately lots of primordial barriers to that being useful.

Putting aside the exception-vs-Builder-error issue, I think the current behavior is intentional. `1.B` is true, `0.B` is false, anything else is illegal.

albert-magyar, 2020-05-04:
I've added a stronger behavior for these "accumulated errors" in #1425 -- they will now immediately throw exceptions if they are detected outside of a "hardware generation context," as when used in a REPL, since it would not make sense to "accumulate" them for a later report.

I think that should solve the odd behavior from the example, so I'll close this issue for now.

closed 2020-05-04 by albert-magyar