chipsalliance/chisel #1423: Please let me assign bit

xzores, 2020-05-03:
class TestModule extends Module {
  val io = IO(new Bundle {
    val in = Input(UInt(10.W))
    val bit = Input(Bool())
    val out = Output(UInt(10.W))
  })
  io.out(0) := io.bit
}

Please make this code valid again, using bundles and vector are to slow to write, when I want to such a simple thing.

sequencer, 2020-05-03:
Unlike traditional HDL language, chisel type system `UInt` is an unsigned int. So it's not rational to assign a bit of an `UInt`, you can explicitly use `Vec(10, Bool())` for this usage, and if you really need to cast it `UInt`, you should explicitly use `asUint`.

xzores, 2020-05-03:
But how do I handle this when I use registers?
Can you tell me how I can make a register where the first bit is an input and the rest is simple some function of the first? 

```
  val myReg = RegInit(0.U(10.W));
  myReg(0) := someInput;
  myReg := myReg << 1;
```
Something like this with a vector?

xzores, 2020-05-03:
Besides I mean you can valid reasons for doing something to a single bit in an UInt.

albert-magyar, 2020-05-03:
> ```
>   val myReg = RegInit(0.U(10.W));
>   myReg(0) := someInput;
>   myReg := myReg << 1;
> ```

If you translated the following into Verilog in the most direct way possible, the part-select assignment would actually end up getting ignored entirely. As I understand it, this is the most direct translation of your code:

```verilog
reg [9:0] myReg;
always @ (posedge clk) begin
   myReg[0] <= someInput;
   myReg <= myReg << 1;
end
```

Could you maybe provide a Verilog sketch of what you specifically want to do? That might be helpful for suggesting Chisel alternatives.

One philosophical difference between Chisel and Verilog is that the built-in "features" of Chisel are more constrained than Verilog in some cases, since the meta-programming features of Chisel generators make it possible to build reusable abstractions that still let you get exactly the microarchitecture you intended. This is part of why part-select assignment is currently not legal in Chisel: it's possible to write a "part-select-assignable word" library in your own code that wraps an underlying vector with convenience methods.

One extra note: there is currently work underway to support emitting Vecs of 1-bit signals as multi-bit regs in Verilog (see freechipsproject/firrtl#1220). This would further incentivize using a Vec.