chipsalliance/chisel #1427: Timing constraints in chisel

sumanth102, 2020-05-05:
can i specify timing constraints for particular pins in chisel?

edwardcwang, 2020-05-05:
Hello,

Thanks for your interest in Chisel! While the Chisel project primarily focuses on design at the RTL level, there is a complementary project, Hammer, that provides physical design abstractions to specify things like pin timing and placement constraints. You can find a reference for its pin APIs below:

https://github.com/ucb-bar/hammer/blob/master/src/hammer-vlsi/defaults.yml

sumanth102, 2020-05-05:
Thank you. this helps a lot. Just so i understand this right, i can use this in conjunction with chisel?
would i be able to implement this on something like a rocket chip?

jwright6323, 2020-05-05:
If you want to build things with rocketchip, chisel, and hammer, you should check out https://github.com/ucb-bar/chipyard

sumanth102, 2020-05-05:
> 
> 
> If you want to build things with rocketchip, chisel, and hammer, you should check out https://github.com/ucb-bar/chipyard

Thanks a lot!

closed 2020-05-19 by azidar

ckmufeng, 2020-08-04:
Hello, I am looking for a hardware languge that can write timing constaints along with the design. Would Chisel like to implement this ?  It would be a great feature for new hardware languge.