chipsalliance/chisel #1435: Pretty Print Failing Line/Column

seldridge, 2020-05-12:
tl;dr: Print out GCC/LLVM-style to pretty-print error information with the failing line and a carat to the failing column.

### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

Consider the following circuit that will error during elaboration:

```scala
class Foo extends MultiIOModule {
  val in = IO(Input(UInt(1.W)))
  val out = IO(Output(SInt(1.W))) 
  out := in
}
```

Currently connecting an `SInt` and `UInt` errors with the correct, but possibly confusing:
```
[info] [0.005] Elaborating design...
[error] chisel3.internal.ChiselException: Connection between sink (SInt<1>(IO in unelaborated Foo)) and source (UInt<1>(IO in unelaborated Foo)) failed @: Sink (SInt<1>(IO in unelaborated Foo)) and Source (UInt<1>(IO in unelaborated Foo)) have different types.
[error] 	...
[error] 	at Playground$Foo.<init>(main.scala:10)
[error] 	at Playground$.$anonfun$new$1(main.scala:13)
[error] 	... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)
```

While this is useful, it's still a hunt ot figure out what's going on. It would be way better to pretty print out some GCC/LLVM-style error reporting. Something like:

```
[error] Connection between sink (SInt<1>(IO in unelaborated Foo)) and source (UInt<1>(IO in unelaborated Foo)) failed @: Sink (SInt<1>(IO in unelaborated Foo)) and Source (UInt<1>(IO in unelaborated Foo)) have different types.
[error] In file Foo.scala:
[error]   out := in
[error]   ~~~~^
```

This should be pretty trivial to do in FIRRTL once source locators are available, but I'm not sure of the implementation to do this during elaboration in Chisel.

### Type of Feature

- documentation                     

- new feature/API

### Related Features

No.

### External Information

None.

azidar, 2020-05-19:
Other areas to pretty-print: type mismatched of fields in bundles. Could report the scala file as well.

jackkoenig, 2023-05-09:
This is done in https://github.com/chipsalliance/chisel/pull/3039.

closed 2023-05-09 by jackkoenig