chipsalliance/chisel #1444: Option to Disable Randomization Logic

seldridge, 2020-05-14:
Currently, there are a number of personal annoyances with how Chisel/FIRRTL emits verbose randomization logic. This issues is a proposal to add an option to disable randomization logic generation (which would be realized in FIRRTL). Additionally, this brings up a broader discussion about the motivation/need for randomization which I wrote up here.

Annoyances with randomization:

1. Randomization clutters up the Verilog for teaching/presentation/marketing purposes and there's no way to turn this off
2. (Tenuous, position:) Simulators support X-initialization. With this support, isn't randomization a parameter that a testing library should be passing to it's backend and not a Verilog generation problem?
3. (Unresearched position:) Verification teams know what they're doing (randomization can be turned off for them) and they would benefit from smaller Verilog files
4. Randomization logic is emitted even when registers are reset

Counterpoints supporting randomization:

1. Existing Verilog generation with randomization logic provides stability across simulators
2. `chisel3.util` contains generators that suffer from X-pessimism meaning that Chisel designs using the standard library may fail to simulate, but be perfectly correct.

At bare minimum, I'd like to have an option to disable randomization logic from being emitted. This shouldn't be the default, just something that a user can turn on if they have some good reason to. Two examples here would be: (1) me writing a Chisel snippet to go on the website and (2) ChiselTest opting to do randomization directly via a Verilator arg.

As further points of discussion:

1. What is the motivation in choosing to randomize in generated Verilog over relying on the simulator (VCS, Verilator, etc.) to do it? Are there deficiencies or inconsistencies across different simulators?
2. If a register is reset, why should it have randomization logic emitted? (Is it problematic to have this be `X` up until it's reset?)
3. Are there any Verification engineers out there who have an opinion on this? Do you rely on randomization logic that Chisel emits?
4. Generator writers should know if a generator artifact needs randomization to work or not. Would it benefit from introducing an annotation that can be used to selectively enable/disable randomization for specific registers?

As a concrete example, consider the following simple Chisel circuit. Assume that I'm somebody who's showing a colleague how to use Chisel:

```scala
class Foo extends MultiIOModule {
  val in = IO(Input(Bool()))
  val out = IO(Output(Bool()))
  out := RegNext(~in)
}
```

This produces the sane high FIRRTL IR:

```
circuit Foo :
  module Foo :
    input clock : Clock
    input reset : UInt<1>
    input in : UInt<1>
    output out : UInt<1>
  
    node _T = not(in)
    reg _T_1 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), _T_1)
    _T_1 <= _T
    out <= _T_1
```

And vomits the following Verilog:

```verilog
module Foo(
  input   clock,
  input   reset,
  input   in,
  output  out
);
  reg  _T_1;
  reg [31:0] _RAND_0;
  assign out = _T_1;
`ifdef RANDOMIZE_GARBAGE_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_INVALID_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_REG_INIT
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_MEM_INIT
`define RANDOMIZE
`endif
`ifndef RANDOM
`define RANDOM $random
`endif
`ifdef RANDOMIZE_MEM_INIT
  integer initvar;
`endif
`ifndef SYNTHESIS
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
  `ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  _T_1 = _RAND_0[0:0];
  `endif // RANDOMIZE_REG_INIT
  `endif // RANDOMIZE
end // initial
`endif // SYNTHESIS
  always @(posedge clock) begin
    _T_1 <= ~in;
  end
endmodule
```

I would like to have an option to get something cleaner:

```verilog
module Foo(
  input   clock,
  input   reset,
  input   in,
  output  out
);
  reg  _T_1;
  assign out = _T_1;
  always @(posedge clock) begin
    _T_1 <= ~in;
  end
endmodule
```

**Type of issue**: feature request

**Impact**: API addition (no impact on existing code) | Verilog generation

**Development Phase**: request

**Other information**
 

None.

**What is the use case for changing the behavior?**

Discussed above.

mwachs5, 2020-05-14:
For point "If a register is reset, why should it have randomization logic emitted? (Is it problematic to have this be X up until it's reset?)"

I agree -- I wasn't even aware this was the case.

albert-magyar, 2020-05-15:
I think the concern is that it will leak X values into non-reset registers and increase the exposure of the simulation to X pessimism and optimism.

It's hard to say whether that would be a meaningful problem given our current strategy of emitting very conservative conditional logic and expressions. However, dropping initializations of reset registers would definitely make it harder to introduce some "better Verilog" patterns like streamlined procedural `if`s/`else`s (which is a goal) without introducing cases where X optimism is observable.

I think this gets to the heart of a lot of our Verilog emission issues: we have multiple layers of safeguards against simulation issues, which are quite often excessively conservative. However, small peephole optimizations often leave us surprised by unexpected outcomes when suddenly some holes in the assumptions align. I think this is a good argument for a clean-sheet look at Verilog emission down the road.

However, for now, I think that making it optional is a win-win.

aswaterman, 2020-05-15:
Removing randomization from registers that are reset will cause problems in
general. For synchronous reset, running the clock to perform the reset will
transfer Xs from registers that are reset into registers that are not
reset. Even for async reset, some parts of the design may be reset at
different times, so Xs lingering in not-yet-reset parts can pessimize other
parts of the design.

My experience relying on simulators to do the randomization has been mixed.
It’s obviously fine if the simulator has a 2-state-simulation option. My
experience using VCS 4-state simulation + using VCS to randomize the
initial state was quite negative; the feature was basically broken.

Having an option to suppress emission of randomization code seems fine;
jettisoning randomization altogether is not gonna work out for us, though.
But, separately, it would be good to improve the way the emitted code looks
for marketing purposes.

azidar, 2020-05-19:
An option to disable that randomization emission seems sane. Let's do it.

closed 2021-11-19 by mergify[bot]

peterhchen, 2025-02-26:
Can you provide a simple example of how to "disable the generate/printout randomization logic", such as, println (new <class name>, disableOption)?

seldridge, 2025-02-26:
Randomization control is an option to the `firtool` FIRRTL compiler, `-disable-all-randomization`. You can access this via the `firtoolOpts` argument to methods defined on `ChiselStage`. E.g., the following is a complete example that will disable all randomization logic:

``` scala
//> using repository "sonatype-s01:snapshots"
//> using scala "2.13.15"
//> using dep "org.chipsalliance::chisel:7.0.0-M2+455-e4297b9e-SNAPSHOT"
//> using plugin "org.chipsalliance:::chisel-plugin:7.0.0-M2+455-e4297b9e-SNAPSHOT"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
// _root_ disambiguates from package chisel3.util.circt if user imports chisel3.util._
import _root_.circt.stage.ChiselStage

class Foo extends Module {
  val d = IO(Input(UInt(8.W)))
  val q = IO(Output(chiselTypeOf(d)))

  private val r = Reg(chiselTypeOf(q))

  r :<= d
  q :<= r
}

object Main extends App {
  println(
    ChiselStage.emitSystemVerilog(
      gen = new Foo,
      firtoolOpts = Array(
        "-disable-all-randomization", // <-- This is the option
        "-strip-debug-info",
        "-enable-layers=Verification",
        "-enable-layers=Verification.Assert",
        "-enable-layers=Verification.Assume",
        "-enable-layers=Verification.Cover"
      )
    )
  )
}
```

This produces the following Verilog:

``` verilog
// Generated by CIRCT firtool-1.107.0
module Foo(
  input        clock,
               reset,
  input  [7:0] d,
  output [7:0] q
);

  reg [7:0] r;
  always @(posedge clock)
    r <= d;
  assign q = r;
endmodule
```

peterhchen, 2025-02-26:
Hi, Seldridge: 

Thanks for the response:
I used Chisel inside the Jupyter notebook.
The code is in the following link:
> jupyter notebook 04.state.py
https://github.com/peterhchen/AIU_CS800_Chisel/tree/main/500_UCSC_HWD/004_SeqCircuit/001_Code

// default Register:
// val r = Reg(Bool())
class RegLand extends Module {
    val io = IO(new Bundle {
        val in  = Input(Bool())
        val en  = Input(Bool())
        val out = Output(Bool())
    })
    val r = Reg(Bool())
//    val r = RegInit(0.B)
    r := io.in
    io.out := r
//     io.out := RegNext(io.in, 0.B)
//     io.out := RegEnable(io.in, 0.B, io.en)
}

// There is an option in CHisel to disable the print out of "randomize Logic" 
// https://github.com/chipsalliance/chisel/issues/1444
println (getVerilog(new RegLand))

Below is the Verilog code:
Elaborating design...
Done elaborating.
module RegLand(
  input   clock,
  input   reset,
  input   io_in,
  input   io_en,
  output  io_out
);
`ifdef RANDOMIZE_REG_INIT
  reg [31:0] _RAND_0;
`endif // RANDOMIZE_REG_INIT
  reg  r; // @[cmd27.sc 7:16]
  assign io_out = r; // @[cmd27.sc 10:12]
  always @(posedge clock) begin
    r <= io_in; // @[cmd27.sc 9:7]
  end
// Register and memory initialization
`ifdef RANDOMIZE_GARBAGE_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_INVALID_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_REG_INIT
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_MEM_INIT
`define RANDOMIZE
`endif
`ifndef RANDOM
`define RANDOM $random
`endif
`ifdef RANDOMIZE_MEM_INIT
  integer initvar;
`endif
`ifndef SYNTHESIS
`ifdef FIRRTL_BEFORE_INITIAL
`FIRRTL_BEFORE_INITIAL
`endif
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
`ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  r = _RAND_0[0:0];
`endif // RANDOMIZE_REG_INIT
  `endif // RANDOMIZE
end // initial
`ifdef FIRRTL_AFTER_INITIAL
`FIRRTL_AFTER_INITIAL
`endif
`endif // SYNTHESIS
endmodule

I want to have an option: 
println (getVerilog(new RegLand) + "-disable-all-randomization")

and produces the following Verilog code without Randomization logic:
module RegLand(
  input   clock,
  input   reset,
  input   io_in,
  input   io_en,
  output  io_out
);
  reg [31:0] _RAND_0;
  reg  r; // @[cmd27.sc 7:16]
  assign io_out = r; // @[cmd27.sc 10:12]
  always @(posedge clock) begin
    r <= io_in; // @[cmd27.sc 9:7]
  end
endmodule

===
How to do that with an example?

Thanks for your time, Peter H. Chen

seldridge, 2025-02-26:
I don't know where that `getVerilog` method is coming from. However, you can use:

``` scala
import circt.stage.ChiselStage

ChiselStage.emitSystemVerilog(new RegLand, firtoolOpts=Array("-disable-all-randomization"))
```

peterhchen, 2025-02-26:
Thanks Eldrige:

Thanks for feedback:
In Jupyter Notebook, I changed below.
===
import circt.stage.ChiselStage
//import chisel3._
//import chisel3.stage._
ChiselStage.emitSystemVerilog(new RegLand, firtoolOpts=Array("-disable-all-randomization"))
=== Error Message ===
cmd3.sc:19: not found: value circt
import circt.stage.ChiselStage
       ^cmd3.sc:22: not found: value ChiselStage
val res3_2 = ChiselStage.emitSystemVerilog(new RegLand, firtoolOpts=Array("-disable-all-randomization"))
             ^cmd3.sc:22: not found: value firtoolOpts
val res3_2 = ChiselStage.emitSystemVerilog(new RegLand, firtoolOpts=Array("-disable-all-randomization"))
                                                        ^Compilation Failed
Compilation Failed

seldridge, 2025-02-26:
It looks like you're on a pretty old version of Chisel. That command should work with Chisel versions >= 3.6.0.

If you are on a version before that, I think the command is:

``` scala
import chisel.stage.ChiselStage
ChiselStage.emitSystemVerilog(new RegLand, args = Array("--emission-options=disableMemRandomization,disableRegisterRandomization"))
```

Note: there is a difference between `chisel.stage.ChiselStage` (which uses the deprecated Scala-based FIRRTL Compiler) and `circt.stage.ChiselStage` (which uses [llvm/circt](github.com/llvm/circt)). Versions before 3.6.0 only supported the former. 3.6.0 supports both. 5.0.0 (which was the version directly after 3.6.0 / there was no version 4) only supports the latter.

peterhchen, 2025-02-27:
Thanks Eldrige:

I install the chisel using the link below:
https://www.chisel-lang.org/docs/installation
> chisel --version 
>> v1.1.0
I google search "how to update chisel 3.6.0"
Modify the build.sbt:
lazy val chiselSettings = Seq(
  name := "chisel3",
  libraryDependencies ++= Seq(
    //"org.scalatest" %% "scalatest" % "3.2.15" % "test",  <=== I remove this line and add the "3.6.0" below
    "edu.berkeley.cs" %% "chisel3" % "3.6.0",
    "org.scalatest" %% "scalatest" % "3.2.15" % "test",
    "org.scalatestplus" %% "scalacheck-1-14" % "3.2.2.0" % "test",
    "com.lihaoyi" %% "upickle" % "2.0.0"
  )

==
> sbt clean build.sbt
==
> chisel --version
>> v1.1.0   

Problem: The version does not change to "3.6.0"

seldridge, 2025-02-28:
I believe that has correctly updated the dependency on Chisel for that project to use 3.6.0. 

Chisel (the hardware description language) doesn't have a command line entry point called `chisel`. That 1.1.0 may be a different program called Chisel?

peterhchen, 2025-02-28:
I google "how to install chisel ucb on Ubuntu"
https://documentation.ubuntu.com/chisel/en/latest/how-to/install-chisel/

It provides three ways to install chisel ucb
1. install binary: version 1.1.0
2. install source
3. install the snap: version 1.1.0
After installation
> which chisel
>> /usr/local/bin/chisel
> chisel version
>> v1.0.0

Do you know where I can find the document/guide for chisel ucb on Ubuntu?

Thanks a lot for reading this message.
Peter H. Chen

peterhchen, 2025-03-07:
1. google search "linux chisel is same as ucb chisel?"

 "Linux Chisel" and "UCB Chisel" refer to different tools with distinct
purposes: Linux Chisel is a tool for creating secure tunnels, while UCB
Chisel is a hardware description language (HDL).
2. I remove the "sudo rm -rf /snap/bin/chisel" and "sudo rm -rf
/usr/local/bin/chisel"
3. How to install UCB chisel?
https://github.com/freechipsproject/chisel-bootcamp/tree/master/source
https://github.com/freechipsproject/chisel-bootcamp/blob/master/source/load-ivy.sc
The load-ivy.sc specified the chisel version.
https://github.com/agile-hw/lectures/blob/main/resource/chisel_deps.sc
The chisel-deps.sc specified chisel version 3.6.0

peterhchen, 2025-03-08:
Below is my find out, you may have encounter the same problems: 
1. In the below link, there is a chisel_deps.sc written by Prof. Sbeamer (Scott Beamer). He defines convenience function getVerilog(). The getVerilog() has  option to "disable Randomization Logic", 

def getVerilog(dut: => chisel3.RawModule): String = {
  val arguments = Array("--emission-options",
                        "disableMemRandomization,disableRegisterRandomization",
                        "--info-mode", "ignore")
	try {
		removeAllComments((new chisel3.stage.ChiselStage).emitVerilog(dut, arguments))
	} catch {
		case e: Exception => s"An exception occurred: ${e.getMessage}"
	}
}

// Convenience function to invoke Chisel and grab emitted FIRRTL.
def getFirrtl(dut: => chisel3.RawModule): String = {
  val arguments = Array("--emission-options",
                        "disableMemRandomization,disableRegisterRandomization",
                        "--info-mode", "ignore")
	try {
		removeAllComments((new chisel3.stage.ChiselStage).emitFirrtl(dut, arguments), " @")
	} catch {
		case e: Exception => s"An exception occurred: ${e.getMessage}"
	}
}

2. However, my sc (Scala Configuration) file is load from UCB Bootcamp, load-ivy.sc with different dependencies. It does not have "--emission-options".

1. Prof. Scott Beamer
https://github.com/agile-hw/lectures/tree/main/resource
2. UCB Chisel Bootcamp:
https://github.com/freechipsproject/chisel-bootcamp/tree/master/source

seldridge, 2025-03-08:
Sorry, yes, there are several projects names Chisel. You can usually narrow your search by looking for "Chisel HDL".

I would encourage you to look at the Chisel installation instructions here: https://www.chisel-lang.org/docs/installation#quickstart-with-scala-cli

The easiest way to start with Chisel is by using Scala CLI and the example that those docs tells you to download via curl. That content is included below:

``` scala
//> using scala "2.13.12"
//> using dep "org.chipsalliance::chisel:6.7.0"
//> using plugin "org.chipsalliance:::chisel-plugin:6.7.0"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
// _root_ disambiguates from package chisel3.util.circt if user imports chisel3.util._
import _root_.circt.stage.ChiselStage

class Foo extends Module {
  val a, b, c = IO(Input(Bool()))
  val d, e, f = IO(Input(Bool()))
  val foo, bar = IO(Input(UInt(8.W)))
  val out = IO(Output(UInt(8.W)))

  val myReg = RegInit(0.U(8.W))
  out := myReg

  when(a && b && c) {
    myReg := foo
  }
  when(d && e && f) {
    myReg := bar
  }
}

object Main extends App {
  println(
    ChiselStage.emitSystemVerilog(
      gen = new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```

Once you install `scala-cli`, you can then run this with, `scala-cli chisel-example.scala`. That will then print Verilog to your terminal. The option to disable randomization is already included.

You can also use the Chisel template instructions to get a basic Chisel project: https://github.com/chipsalliance/chisel-template

peterhchen, 2025-03-10:
Hi Eldridge, Thanks.

*I follow your guide and reinstalled: *
https://www.chisel-lang.org/docs/installation

*The error messages are still the same. *
*UCSC Professor Scott Beamer defines his sc (scala configuration) *
https://github.com/agile-hw/lectures/tree/main/resource/chsiel-deps.sc
His dependencies are very complicated. The chisel-deps.sc defined the
option ("...". "...skip random logic ...", ...) in Array().

The one I used is from UC Berkeley Bootcamp.
https://github.com/freechipsproject/chisel-bootcamp/tree/master/source/load-ivy.sc
There is no error message for most of the case. Only a few examples from
Professor Scott Beamercannot run.
In the load-ivy.sc getVerilog() does not define the "skip random logic" in
Array().

Is there a chisel document link which describes how to define the "skip
random logic".

Thanks for reading my question.

Peter H. Chen