chipsalliance/chisel #1445: Chisel Integration is Painful

johnsbrew, 2020-05-19:
**Type of issue**:  other enhancement
**Impact**: Unknown
**Development Phase**: discussion

_I started writing this into a comment of #1361 but i figured out that a proper issue would be more appropriate for community discussion_

### Chisel integration is really painful
- the base fact that most chisel objects must be manipulated in builder context make everything difficult as soon as one is not actually emitting a module : most simple example is that one cannot retrieve the name of a module without inspecting elaboration result
- ChiselStage entry points are limited and not quite documented
- As soon as one leave the standard intented flow of Chisel as master, everything gets complicated
- the flow is slow (long starting time for each steps: elab, firrtl compilation , ...) and is getting slower & slower: another half second added for each firrtl compilation start-up with the new Stage/Phase system

I know that all these limitations originate from scala but I wonder how far we could make integration smoother in order to reach a satisfactory user experience: it is currently not ready to be presented to any hardware engineer without providing a proper support team and I hope one day the support team requirement could be dropped :)

Just to give you some insights, we started looking at chisel 2 years ago and we are just about to have a first dummy module actually integrated in production. Despite not a full time job for us, this painful integration demonstrates how far chisel is from the "of-the-shelf production-ready" solution it aims at being. And once again I really wish it will become it one day, because I definitely do not want to start again designing with verilog or vhdl for a living!

Of course I will be glad to share at some point more details and contribute to better integrability, in particular tools we developed to integrate chisel into larger SystemVerilog codebase.

In a nutshell I think we need to create the lib that would be to chisel what ammonite is to scala(-repl).

I would be glad to collect your thoughts about all that, as you will probably point out other flaws that are outside of my current scope. I hope this could lead to interesting contributions!
Ping @jackkoenig @seldridge @azidar @edwardcwang @ducky64 @chick @mwachs5 @schoeberl

azidar, 2020-05-19:
We are brainstorming ideas, and had a few thoughts.

1) What are the Chisel Stage/Phase API's that you are looking for?
2) What do you think about having Chisel/FIRRTL packages, where they are able to elaborate to Verilog independently, but refer/import each other in the Chisel. This enables more incremental compilation approach, as well as force us to think outside of a single Builder context.

We hear you, and we've experienced similar struggles. To move the conversation forward, do you have some concrete use-cases which help clarify the nature of your integration problems?

jackkoenig, 2020-05-19:
On the performance side; this has been of particular annoyance to me lately. I hope to eventually have a full, shared benchmarking flow to prevent performance regressions. You'll notice that I have made several PRs in the last week related to performance: https://github.com/freechipsproject/firrtl/pull/1475, https://github.com/freechipsproject/firrtl/pull/1582, https://github.com/freechipsproject/firrtl/pull/1601, and https://github.com/freechipsproject/firrtl/pull/1602 (more to come). I hope to eventually get to larger scale improvements like parallelization and incremental compile; but even without that, as Adam mentions above, we can give the designer more control over splitting up their designs themselves which would help on this as well.