chipsalliance/chisel #1456: Vec generates unreasonably large verilog with unnecesarry wires

arun13e, 2020-05-30:
We are creating a 16 core chip, it is a heavily modified version of rocket-chip, the size of the verilog with one core is 10 MB, the size of 16 cores is over 1 GB.
It is logistically unworkable as it takes days to compile the verilog.

We isolated the issue to the way that chisel handles Vec. Vec of 64 bit width will generate 64 separate wires and any operation performed with that Vec will require assign statement for all 64 of them. An improvement would be for Vec to generate a single verilog Wire of 64 bit. `wire myVec[63:0]` 

For eg We are using original rocket TLDebug module 

**4 cores 100 MB verilog in which
20%  is TLDebugModuleInner : 2,33,034‬ lines, 9259 cells**

**2 cores 44 MB verilog in which
9%  TLDebugModuleInner : 66,595 lines, 8556 cells**

The following zip contains output verilog as well as the chisel code for the above mentioned 

[VEC.zip](https://github.com/freechipsproject/chisel3/files/4706185/VEC.zip)

jackkoenig, 2020-05-30:
Thank you for taking the time to provide this feedback, and thank you for digging into the cause. Vecs are certainly in need of improvement, and I hope we are able to get to it soon.

Such long compilation times are unacceptable, so until we have fixed the underlying issue, let me suggest a few things you can do to minimize the pain for the time being:

(This ended up being a lot longer than I expected, but hopefully it's useful for dealing with such issues)

Looking at `TLDebugModuleInner` (https://github.com/chipsalliance/rocket-chip/blob/7a8b3679d8aed1553c466cdf09fc546fb08f1349/src/main/scala/devices/debug/Debug.scala#L671), there are tons of `Vec`s of `Bool` which I suspect are causing the bulk of the compilation issues. I suggest replacing each one of these `Vec`s of `Bool` with a `UInt` of the same size. It is a fairly simple transformation that can have an enormous impact on the compilation speed and size of the emitted Verilog.

Let us consider an example from `TLDebugModuleInner`. I've taken one of these `Vec` of `Bool` and transformed a few of it's uses:
```scala
val haltedBitRegs    = Reg(Vec(nComponents, Bool()))
```
Becomes
```scala
val haltedBitRegs    = Reg(UInt(nComponents.W))
```
#### Example 1
```scala
DMSTATUSRdData.anyhalted := ((~io.debugUnavail &  haltedBitRegs) & hamaskFull).reduce(_ | _)
```
This is a very simple modification. Most of these operations being used are defined on class `UInt` to operate at the world-level. The only change necessary is the Scala Seq `.reduce` is operating a bit-level on the Vec, instead we use the Chisel built-in `.orR` (or reduction). We also do need to change each other `val` involved to a `UInt`.
```scala
DMSTATUSRdData.anyhalted := ((~io.debugUnavail &  haltedBitRegs) & hamaskFull).orR
```
#### Example 2
```scala
haltedStatus(ii) := Cat(haltedBitRegs.slice(ii * 32, (ii + 1) * 32).reverse)
```
This one is a little more complicated. One important observation is that `Cat(myVec.reverse)` is actually equivalent to `myVec.asUInt`, so we can just remove those, but we're left with `slice` which is a [Scala `Seq` operation](https://www.scala-lang.org/api/2.12.10/scala/collection/Seq.html#slice(from:Int,until:Int):Repr). In this context, it is essentially selecting a 32 Bools from the Vec. Put another way, it's selecting a 32-bit slice from the full word:
```scala
haltedStatus(ii) := (haltedBitRegs >> (ii * 32))(31, 0) // note may need to deal with < 32-bit result
```
#### Example 3
These are the easy examples, but what about the main use case in Chisel for `Vec` over `UInt`: dynamic indexing? This transformation is a little more complicated. The idea is that you create some logic to construct a full-sized UInt result, and connect that to the Reg. For `Vec` of `Bool` it's easier because you can think about it as masking.
```scala
    for (component <- 0 until nComponents) {
      when (~io.dmactive || ~dmAuthenticated) {
        haltedBitRegs(component) := false.B
      }.otherwise {
        // Hart Halt Notification Logic
        when (hartIsInResetSync(component)) {
          haltedBitRegs(component) := false.B
        }.elsewhen (hartHaltedWrEn) {
          when (hartSelFuncs.hartIdToHartSel(hartHaltedId) === component.U) {
            haltedBitRegs(component) := true.B
          }
        }.elsewhen (hartResumingWrEn) {
          when (hartSelFuncs.hartIdToHartSel(hartResumingId) === component.U) {
            haltedBitRegs(component) := false.B
          }
        }
```
I personally find this logic really hard to follow, a rewrite might make it more clear (and more efficient), so here's an attempt:
```scala
  when(~io.dmactive || ~dmAuthenticated) {
    haltedBitRegs := 0.U
  }.otherwise {
    // Summarizing this chunk of logic in English
    // For each bit
    //   * if hartIsInResetSync is true, clear it
    //   * else if hartHaltedWrEn, set bit of hartHaltedId
    //   * else if hartResumingWrEn, clear bit of hartResumingId

    // Low priority default that we'll conditionally override
    val haltedBitRegsNext = WireInit(haltedBitRegs)
    
    // Hart Halt Notification Logic is higher priority than what follows so we'll put it last for last connect semantics
    // Note that its when condition is per-bit while the later are
    val hartHaltedIdIndex = UIntToOH(hartSelFuncs.hartIdToHartSel(hartHaltedId))
    val hartResumingIdIndex = UIntToOH(hartSelFuncs.hartIdToHartSel(hartResumingId))

    //  else if hartHaltedWrEn, set bit of hartHaltedId
    when(hartHaltedWrEn) {
      haltedBitRegsNext := haltedBitRegs | hartHaltedIdIndex
    // else if hartResumingWrEn, clear bit of hartResumingId
    }.elsewhen(hartResumingWrEn) {
      haltedBitRegsNext := haltedBitRegs & ~hartResumingIdIndex
    }
    // Hart Halt Notification Logic (clear each bit with corresponding bit in hartIsInResetSync set)
    haltedBitRegs := haltedBitRegsNext & ~hartIsInResetSync
  }
}
```
```

arun13e, 2020-07-16:
The solution you wrote worked out great although at first it seems like introducing extra logic of applying masks to large width wires, but after I synthesized the output verilog to a physical design, the results are more or less the same in terms and area and timing, Thank you for the help.

closed 2023-05-09 by jackkoenig