chipsalliance/chisel #1465: Bad error reporting on wrong (subword) assignments

crepererum, 2020-06-06:
**Type of issue**: other enhancement

**Impact**: no functional change

**Development Phase**: request

**Other information**
This is more a meta-issue mostly related to the beginners user experience. I think most of these issues are not done by more experienced devs.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

1. Start with [chisel-template](https://github.com/freechipsproject/chisel-template/) (commit `11f6ca470120908d167cb8dc3241953eb31d0acb` was used for this experiment)
2. Create a new file in `src/main/scala/gcd/Bug.scala` (see content below)
3. Run `sbt "runMain gcd.BugDriver"` to reproduce the output.
4. See how the error reporting explain (or does not explain) the bad assignment.

**What is the current behavior?**

These are examples where the error reporting is (IMHO) bad / misleading.

---
*Wrong transfer direction:*

```scala
package gcd

import chisel3._
import chisel3.stage.{ChiselGeneratorAnnotation, ChiselStage}

class BugModule extends Module {
  val io = IO(new Bundle {
    val in  = Input(UInt(16.W))
    val out = Output(UInt(16.W))
  })

  io.in := io.out
}

object BugDriver extends App {
  (new ChiselStage).execute(args, Seq(ChiselGeneratorAnnotation(() => new BugModule)))
}
```

```text
[info] Loading settings for project chisel_bad_error-build from plugins.sbt ...
[info] Loading project definition from /home/tom/src/bugs/chisel_bad_error/project
[info] Loading settings for project chisel_bad_error from build.sbt ...
[info] Set current project to chisel-module-template (in build file:/home/tom/src/bugs/chisel_bad_error/)
[info] Compiling 1 Scala source to /home/tom/src/bugs/chisel_bad_error/target/scala-2.12/classes ...
[warn] there were two feature warnings; re-run with -feature for details
[warn] one warning found
[info] Done compiling.
[info] running gcd.BugDriver 
[[35minfo[0m] [0.002] Elaborating design...
[[31merror[0m] chisel3.internal.ChiselException: Connection between sink (UInt<16>(IO in unelaborated BugModule)) and source (UInt<16>(IO in unelaborated BugModule)) failed @: Sink is unwriteable by current module.
[[31merror[0m] 	...
[[31merror[0m] 	at gcd.BugModule.<init>(Bug.scala:14)
[[31merror[0m] 	at gcd.BugDriver$.$anonfun$new$1(Bug.scala:19)
[[31merror[0m] 	... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)
[error] (run-main-0) firrtl.options.StageError: 
[error] firrtl.options.StageError: 
[error] 	at chisel3.stage.ChiselStage.run(ChiselStage.scala:49)
[error] 	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error] 	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:37)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Translator.transform(Phase.scala:240)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:240)
[error] 	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:46)
[... 460 lines cut ...]

```text
[info] Loading settings for project chisel_bad_error-build from plugins.sbt ...
[info] Loading project definition from /home/tom/src/bugs/chisel_bad_error/project
[info] Loading settings for project chisel_bad_error from build.sbt ...
[info] Set current project to chisel-module-template (in build file:/home/tom/src/bugs/chisel_bad_error/)
[info] running gcd.BugDriver 
[[35minfo[0m] [0.002] Elaborating design...
[[35minfo[0m] [0.097] Done elaborating.
Computed transform order in: 397.3 ms
[error] (run-main-0) firrtl.passes.CheckWidths$BitsWidthException:  @[Bug.scala 14:18]: [target ~BugModule|BugModule] High bit 16 in bits operator is larger than input width 16 in bits(io.in, 16, 0).
[error] firrtl.passes.CheckWidths$BitsWidthException:  @[Bug.scala 14:18]: [target ~BugModule|BugModule] High bit 16 in bits operator is larger than input width 16 in bits(io.in, 16, 0).
[error] stack trace is suppressed; run 'last Compile / bgRunMain' for the full output
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 4 s, completed Jun 6, 2020 2:02:18 PM
```

---

**Please tell us about your environment:**
See linked chisel-template commit.

**What is the use case for changing the behavior?**
New-user experience.

seldridge, 2020-06-06:
Thanks for taking the time to report this! :+1: I think these can be cleaned up. That `requirement failed` error is pretty egregious.

One clarification: is the concern here just the content of the error messages or is it also anything related to the stack trace? Running this locally, parts of that stack trace that you get is hidden for me, but not all of it.

One question: do you have any suggestions on what you would like to see here?

Ideally, I'd like to get to something like GCC/LLVM-style pretty printing of errors: https://github.com/freechipsproject/chisel3/issues/1435.

One note: the "good" error message is actually coming from the FIRRTL compiler. We have a little more control there in terms of generating the message which helps. There's been discussions about further improving this with source maps to enable the actual original code to be shown.

crepererum, 2020-06-07:
> is the concern here just the content of the error messages or is it also anything related to the stack trace?

Depends:

- **Information flooding** (*wrong transfer direction, negative bounds*): Too many stack traces so it's hard to spot the actual user error. You have mentioned that you get a shorter output locally and I don't know why this is. I thought the `chisel-template` (at the mentioned commit) should more or less equalize these differences.
- **Error message** (*flipped bounds*): The error message is pretty generic (`java.lang.IllegalArgumentException: requirement failed`) which makes it unclear what the root cause is.

Some background: It all started with the *flipped bounds* issue and I first thought this is an internal compiler error. Then I found the root-cause-line somewhere in the long stacktraces and even later (after fixing the issue) I saw that there is a shorter backtrace at the top of the message.

If both points are addressed and the output looks like the *high bit out of bounds* example, I think this would help new user massively. #1435 would be something on top of it and would be great, as it is something that IMHO people coming from other languages / toolchains (or even just getting started with hardware design, FPGAs and co) would find "normal".

seldridge, 2020-06-07:
> You have mentioned that you get a shorter output locally and I don't know why this is. I thought the chisel-template (at the mentioned commit) should more or less equalize these differences.

I see exactly what you see on `chisel-template`. Sorry for the confusion.

I ran this yesterday in an `sbt console` inside the `chisel3` development repo and got slightly different behavior. It's still printing the stack trace, but it's not quite as much:

```
scala> (new ChiselStage).execute(Array.empty, Seq(ChiselGeneratorAnnotation(() => new BugModule)))
[info] [0.002] Elaborating design...
[error] chisel3.internal.ChiselException: Connection between sink (UInt<16>(IO in unelaborated BugModule)) and source (UInt<16>(IO in unelaborated BugModule)) failed @: Sink is unwriteable by current module.
[error] 	...
[error] 	at $line7.$read$$iw$$iw$$iw$$iw$BugModule.<init>(<console>:21)
[error] 	at $line11.$read$$iw$$iw$$iw$$iw$.$anonfun$res3$1(<console>:17)
[error] 	... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)
firrtl.options.StageError:
  at chisel3.stage.ChiselStage.run(ChiselStage.scala:49)
  at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
  at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
  at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:37)
  at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
  at firrtl.options.Translator.transform(Phase.scala:240)
  at firrtl.options.Translator.transform$(Phase.scala:240)
  at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
  at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:46)
  at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
  at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
  at scala.collection.immutable.List.foldLeft(List.scala:89)
  at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:46)
  at logger.Logger$.$anonfun$makeScope$2(Logger.scala:168)
  at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
  at logger.Logger$.makeScope(Logger.scala:166)
  at firrtl.options.Stage.transform(Stage.scala:46)
  at firrtl.options.Stage.execute(Stage.scala:57)
  ... 36 elided
Caused by: chisel3.internal.ChiselException: Connection between sink (UInt<16>(IO in unelaborated BugModule)) and source (UInt<16>(IO in unelaborated BugModule)) failed @: Sink is unwriteable by current module.
  at chisel3.internal.throwException$.apply(Error.scala:85)
  at chisel3.Data.connect(Data.scala:406)
  at chisel3.Data.$colon$eq(Data.scala:494)
  ... 53 more
```

This is a red herring, though... The issue is still the stack trace.

One thing that you can do: use `ChiselMain` directly instead of constructing your own `App` / main object, e.g., use `runMain chisel3.stage.ChiselMain --module gcd.BugModule` to get the following:

```
sbt:chisel-module-template> runMain chisel3.stage.ChiselMain --module gcd.BugModule
[info] running chisel3.stage.ChiselMain --module gcd.BugModule
[info] [0.001] Elaborating design...
[error] chisel3.internal.ChiselException: Connection between sink (UInt<16>(IO in unelaborated BugModule)) and source (UInt<16>(IO in unelaborated BugModule)) failed @: Sink is unwriteable by current module.
[error] 	...
[error] 	at gcd.BugModule.<init>(BugModule.scala:12)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] 	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-7"
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 1 s, completed Jun 7, 2020, 2:57:49 PM
```

The pruned stack trace here is actually a fake one. When Chisel hits a `ChiselException` (indicating user error), it will print a pruned stack trace and throw a `StageError`. The latter indicates that the stage that threw it has provided information to the user about what happened and can't continue. `ChiselMain` will then catch the `StageError` and just exit so no stack trace is printed.

An alternative, existing solution would be to copy this behavior or to call `ChiselMain.main` instead of using `ChiselStage`. The app could look like the following:

```scala
object BugDriver extends App {
  try {
    (new ChiselStage).execute(args, Seq(ChiselGeneratorAnnotation(() => new BugModule)))
  } catch {
    case _: firrtl.options.StageError => System.exit(1)
  }
}
```

If you then run this you get the stack trace pruned (because the app is catching the `StageError`:

```
sbt:chisel-module-template> runMain gcd.BugDriver
[info] running gcd.BugDriver 
[info] [0.002] Elaborating design...
[error] chisel3.internal.ChiselException: Connection between sink (UInt<16>(IO in unelaborated BugModule)) and source (UInt<16>(IO in unelaborated BugModule)) failed @: Sink is unwriteable by current module.
[error] 	...
[error] 	at gcd.BugModule.<init>(BugModule.scala:12)
[error] 	at gcd.BugDriver$.$anonfun$new$1(BugModule.scala:18)
[error] 	... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-a"
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 1 s, completed Jun 7, 2020, 3:15:28 PM
```

**All this is orthogonal to a better error message, though!** That needs to be fixed.

crepererum, 2020-06-08:
I didn't know about `runMain chisel3.stage.ChiselMain --module gcd.BugModule`. Both [the cookbook](https://www.chisel-lang.org/chisel3/faqs.html#get-me-verilog) and [this stackoverflow question](https://stackoverflow.com/questions/40470153/is-there-a-simple-example-of-how-to-generate-verilog-from-chisel3-module) somewhat are using apps. That said, I find the `ChiselMain` great!

seldridge, 2020-06-08:
This was discussed in the developer meeting today with two proposals put forward:

1. One or more new methods should be added to `ChiselMain` to allow for users to pass annotations
2. Add the ability for passing module construction parameters (for at least `Int` and `String`) on the command line to enable, e.g., `runMain chisel3.stage.ChiselMain --module gcd.BugModule --parameters 8,"foo"`. Automatic serialization libraries, like json4s may provide serialization/deserialization abilities.

kele14x, 2021-11-26:
I search for documents and finally found usage of `ChiselMain` here. I suggest add this to the tutorial documents.