chipsalliance/chisel #1466: Encountered an exception

mwachs5, 2020-06-07:
**Type of issue**: bug report 

**Impact**: unknown

**Development Phase**: request

**Other information**
 

Encountering this error during chisel->firrtl elaboration, would hope to get a more useful error message. I am sure I am doing something wrong but the error message isn't helping me diagnose. Will continue to dig into the problem and update the ticket if I can point what the user error was that led to this case:

```
[error] (run-main-0) java.util.NoSuchElementException: None.get
[error] java.util.NoSuchElementException: None.get
[error] 	at scala.None$.get(Option.scala:529)
[error] 	at scala.None$.get(Option.scala:527)
[error] 	at chisel3.internal.HasId.getRef(Builder.scala:119)
[error] ocketchiat chisel3.internal.HasId.getRef$(Builder.scala:119)
[error] 	at chisel3.Data.getRef(Data.scala:276)
[error] 	at chisel3.internal.firrtl.Converter$.convert(Converter.scala:46)
[error] 	at chisel3.internal.firrtl.Converter$.convert(Converter.scala:50)
[error] 	at chisel3.internal.firrtl.Converter$.convertSimpleCommand(Converter.scala:115)
[error] 	at chisel3.internal.firrtl.Converter$.rec$1(Converter.scala:163)
[error] 	at chisel3.internal.firrtl.Converter$.convert(Converter.scala:201)
[error] 	at chisel3.internal.firrtl.Converter$.convert(Converter.scala:266)
[error] 	at chisel3.internal.firrtl.Converter$.$anonfun$convert$5(Converter.scala:273)
[error] 	at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:238)
[error] 	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 25 s, completed Jun 7, 2020 1:48:54 PM
```

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

clone the `firrtl-error` branch of rocket-chip: https://github.com/chipsalliance/rocket-chip/tree/firrtl-error

```
<usual rocket-chip setup stuff according to its README>
cd vsim
make verilog
```

**What is the current behavior?**

I get a None.get exception

**What is the expected behavior?**

Something detects that the None.get will fail and gives a more useful error message to help identify the user error.

**Please tell us about your environment:**

Chisel3 commit:

```
21ea734 - (7 weeks ago) Merge branch 'master' into 3.3.x - Jim Lawson (HEAD)
```
**What is the use case for changing the behavior?**

Better user experience

jackkoenig, 2020-06-08:
The infamous `None.get` during the `Converter`. This seems to be a common failure mode for a host of issues where a mutable value Chisel expects to get set never does. Here's an example of using `protected val` in `Bundles` which doesn't work right:
https://scastie.scala-lang.org/a3AjCamwTFS5RNv3LEAaRA

mysoreanoop, 2021-03-30:
Is there any smart way of localizing the issue other than incrementally reconstructing the entire code base?

jackkoenig, 2021-03-30:
> Is there any smart way of localizing the issue other than incrementally reconstructing the entire code base?

Unfortunately, the only way I know to deal with it is by hacking on the chisel3 source. It would be nice if Chisel would dump more information by default, but every time this comes up we usually just fix the proximal cause and hope that that is "the last one."

What version of Chisel are you using?

mysoreanoop, 2021-03-30:
chisel3. Taken alone it compiles just fine. But when integrating with Chipyard causes this error. Does the error have anything to do with the chisel version?

jackkoenig, 2021-03-30:
I'll note that most of Chipyard is using the Chisel2 compatibility mode but still uses Chisel 3. Chipyard 1.3 uses Chisel 3.3, Chipyard 1.4 uses Chisel 3.4. It's possible that bumping the Chisel minor version to the newest release for the given minor release could bring in a fix for this issue. Chisel 3.4.2 is the newest that'll work with Chipyard 1.4, Chisel 3.3.3 is the newest for Chipyard 1.3.

Is your code open-source or do you have a small, reproducible example?

mysoreanoop, 2021-03-31:
It's pretty big, and not open-source; we tested it elsewhere and are now integrating with Rocketchip. I will try upgrading my chisel version appropriately, and get back. Thanks.

jackkoenig, 2021-04-23:
Sorry for the delay and pain here @mysoreanoop, I've created a PR that at least provides more context to such issues: https://github.com/chipsalliance/chisel3/pull/1878

It won't fix whatever bug you're hitting but would at least let you know what line to look at which can help for reporting a bug in Chisel and for working around it.

mysoreanoop, 2021-04-26:
Much appreciated!