chipsalliance/chisel #1477: I am newbee here. I want to find a IDE which can "ctrl + click" to jump the define code

xinlnix, 2020-06-17:
Just want to find IDE

sequencer, 2020-06-17:
Intellij IDEA may be good for you

ducky64, 2020-06-17:
For reference, of the chisel developers who use IDEs, most (if not all?) use IntelliJ. It has a Scala plugin which is pretty good.

danielkasza, 2020-06-25:
Visual Studio Code with the "Scala (Metals)" plugin works for me so far. IntelliJ is also good, but I like the vscode interface more.

seldridge, 2020-06-25:
Expanding on @danielkasza's suggestion: any editor that supports Scala metals should work. And it looks like most editors support "go to definition": https://scalameta.org/metals/docs/editors/overview.html. I'd grab one of those, install metals and see where it gets you.

I'll go ahead and close this as I think this is sufficiently answered. In the future, questions like this would probably be better targeted at [StackOverflow using the `[chisel]` tag](https://stackoverflow.com/questions/tagged/chisel) or brought up in the [Gitter Channel](https://gitter.im/freechipsproject/chisel3) or it's [Matrix bridge](https://riot.im/app/#/room/#chisel:matrix.org). It's all good, though! :+1: Thanks for your interesting in getting started with Chisel.

closed 2020-06-25 by seldridge