chipsalliance/chisel #1497: Using "Flipped" on singleton Bundle results in unexpected behavior

danielkasza, 2020-06-24:
**Type of issue**: bug report

**Impact**:  unknown

**Development Phase**: request

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

```scala
import chisel3._
import chisel3.stage.ChiselStage

object CicaBundle extends Bundle {
  val cica = Output(Bool())
}

class Foo extends Module {
  val io = IO(new Bundle {
    val in  = Flipped(CicaBundle)
    val out = Output(Bool())
  })

  // io.in.cica should be an Input, but it is an Output
  io.in.cica := true.B
  
  io.out := ~io.in.cica
}

println((new ChiselStage).emitVerilog(new Foo))
```

**What is the current behavior?**

Chisel emits verilog, but `io.in.cica` is not flipped.

```verilog
module Foo(
  input   clock,
  input   reset,
  output  io_in_cica,
  output  io_out
);
  assign io_in_cica = 1'h1; // @[main.scala 17:14]
  assign io_out = ~io_in_cica; // @[main.scala 19:10]
endmodule
```

**What is the expected behavior?**

Since a singleton Bundle cannot be flipped, this should fail with a helpful message before outputting any FIRRTL or Verilog code.

**Please tell us about your environment:**

Version: `3.3.0`

**What is the use case for changing the behavior?**

As a beginner, I found this behavior very confusing.

yqszxx, 2020-07-16:
```scala
import chisel3._
import chisel3.stage.ChiselStage

class CicaBundle extends Bundle {
  val cica = Output(Bool())
}

class Foo extends Module {
  val io = IO(new Bundle {
    val in  = Flipped(new CicaBundle)
    val out = Output(Bool())
  })

  // io.in.cica is now an input so cannot be driven
  // io.in.cica := true.B
  
  io.out := ~io.in.cica
}

println((new ChiselStage).emitVerilog(new Foo))
```

It seems you need a new instance every time using bundle as a member of another bundle.

Now it works as expected.

```verilog
module Foo(
  input   clock,
  input   reset,
  input   io_in_cica,
  output  io_out
);
  assign io_out = ~io_in_cica; // @[main.scala 16:10]
endmodule
```

sequencer, 2020-07-16:
From the internal side of chisel, I think we should add macro or compiler plugin to throw an exception when deactivating usage like this.
Since Chisel need a runtime `Builder`, and always creating new instance for binding. `object` will be created at first when program start, and have no duplications, So it will certainly causes a lot of troubles like this. I think the question is how to throw this, rather than how to fix this.

jackkoenig, 2020-07-17:
Yeah the solution is definitely to use a `class`, but we certainly should catch this case and report a good error message to the user.

yqszxx, 2020-07-17:
> Yeah the solution is definitely to use a `class`, but we certainly should catch this case and report a good error message to the user.

Let me see if I can fix this😀