chipsalliance/chisel #1510: Bundle asUInt ignores structural width

ducky64, 2020-07-13:
It appears bundle packing behavior ignores the structural (defined in the Bundle def) width, at least for bundle literals. When defining a bundle literal, the .asUInt returns a different value depending on if the literal components have an explicitly specified width or not. See example below.

Potentially related to FIRRTL (I think?) not having a concept of a bundle type?

**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 
Discovered during implementation of #1280

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

This example reproduces the behavior: https://scastie.scala-lang.org/MYZTNgqTQdOrJuUDiuV1vw
```scala
import chisel3._
import chisel3.experimental.BundleLiterals._
import chiseltest._

class LongBundle extends Bundle {
  val a = UInt(32.W)
  val b = UInt(32.W)
}

class Foo extends MultiIOModule {
  val bun1 = (new LongBundle).Lit(_.a -> 0xBE.U, _.b -> 0xEF.U)
  val bun2 = (new LongBundle).Lit(_.a -> 0xBE.U(32.W), _.b -> 0xEF.U(32.W))
  chisel3.printf("bun1=%x, bun2=%x\n", bun1.asUInt(), bun2.asUInt())
}

RawTester.test(new Foo) { c =>
  c.clock.step(1)
}
```

**What is the current behavior?**

The value of bun1 is different from bun2, as shown by the print:
```
bun1=0beef, bun2=0000000be000000ef
bun1=0beef, bun2=0000000be000000ef
```

**What is the expected behavior?**
Both should be equal, namely bun1 should be be000000ef (optionally with leading zeros)

**Please tell us about your environment:**

Scastie, but reproduces on Windows.

**What is the use case for changing the behavior?**

sequencer, 2020-07-13:
I think the main problem comes from 
https://github.com/freechipsproject/chisel3/blob/ac3ad980dd2e5abd43f3b36ecb1fb95b626aa398/core/src/main/scala/chisel3/Aggregate.scala#L539

In this case, problem is that we actually create a lit with unknown width and didn't check the width equality: 
https://github.com/freechipsproject/chisel3/blob/ac3ad980dd2e5abd43f3b36ecb1fb95b626aa398/core/src/main/scala/chisel3/Aggregate.scala#L543
`litArg` fall back to unknown width, which will ask for a minimal width to create the `LitArg`

So a quick fix is to get width from `field` reflectively, if the `!litArg.widthArg.known`, but more question comes. if it's originally a unknown width and need to be infered by FIRRTL?

sequencer, 2020-07-14:
Here are 3 methods to do this job:
1. During elaborating, change behavior at 
https://github.com/freechipsproject/chisel3/blob/9dda7f51dc7e515cec6e1d3445370df607652958/core/src/main/scala/chisel3/Data.scala#L325
"reflecting" width of Data, bind a new `ElementLitBinding` to `Data`.
2. like #1513 did, only fix bundle `Lit` method.
3. fix this at `InferWidth` FIRRTL transform.

Which method one should I adopt? I personally prefer 1 and 3.

ducky64, 2020-08-07:
There's definitely a bunch of quick fixes for this, but what would a proper fix look like?
Would #921 solve this cleanly, and is it worth reviving that effort?