chipsalliance/chisel #1521: Some recommendations on stop

davidmlw, 2020-07-22:
1. Add `stop(code: UInt): Unit`. I am not sure whether it is possible. It could be more useful if `stop` can pass the internal circuit state outside.
2. Add `finish()`, finish has different meanings other than stop. A user-intended finish is very useful for simulation control.

https://github.com/freechipsproject/chisel3/blob/473a13877c60ba9fb13de47542a8397412c2b967/core/src/main/scala/chisel3/Assert.scala#L80-L92

chick, 2020-07-22:
I think this is how stop works. Stop has a return code. Zero is considered success any other value is considered a failure diagnostic code. I think the problem is that the stop result is not exposed by the unit test frameworks. That should not be hard to fix if that is the problem

davidmlw, 2020-07-24:
What is the current verilog emission result? I just have a look. It seems `stop` is translated to "$fatal" and returns no code. Maybe it needs a Chisel3, Firrtl, Treadle synergy to have a standard spec for `stop` and `finish`.

davidmlw, 2020-07-24:
According to https://literature.cdn.keysight.com/litweb/pdf/ads2004a/pdf/verilogaref.pdf and https://verilog.renerta.com/mobile/source/vrg00042.htm. Verilog `$finish` is not used to return status code to system.

So this recommendation need further considerations

sequencer, 2020-07-27:
Maybe the ultimate way to this is trying to add `DPI` calling to chisel?