chipsalliance/chisel #1523: Building chisel locally is broken

nullobject, 2020-07-27:
**Type of issue**: bug report

**Other information**

A [recent pull request](https://github.com/freechipsproject/chisel3/pull/1499) broke the ability to build chisel locally, using the steps defined in the README file.

The changes introduced in the "Basic model checking API" PR rely on updates to the firrtl library, however these changes aren't yet published in the latest firrtl snapshot.

This means that the build will pass on CI, because it locally publishes the firrtl library. But anyone trying to build the project on their own machine will see compilation errors.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

    $ sbt build

**What is the current behavior?**

```
Error:(130, 35) object Formal is not a member of package firrtl.ir
        case Formal.Assert => fir.Formal.Assert
Error:(131, 35) object Formal is not a member of package firrtl.ir
        case Formal.Assume => fir.Formal.Assume
Error:(132, 34) object Formal is not a member of package firrtl.ir
        case Formal.Cover => fir.Formal.Cover
Error:(134, 16) object Verification is not a member of package firrtl.ir
      Some(fir.Verification(firOp, convert(info), convert(clk, ctx),
```

edwardcwang, 2020-07-27:
Did you also build and publishLocal the master branch of FIRRTL?

albert-magyar, 2020-07-27:
You should also `publishLocal` the master branch of Treadle.

The overall setup to run tests for development work on Chisel:
1. `publishLocal` current working copy of FIRRTL
2. `publishLocal` current working copy of Treadle
3. `sbt test` in chisel3

chick, 2020-07-27:
New snapshots need to be published, I am working on getting that done today

jackkoenig, 2020-07-27:
We need to update the README to reflect the dependency on Treadle as well as to clarify the whole `publishLocal` process, especially because newer snapshots will override your local publishing. It would also be helpful to highlight SBT commands to help see what versions of things are being used.

nullobject, 2020-07-27:
> We need to update the README to reflect the dependency on Treadle as well as to clarify the whole publishLocal process, especially because newer snapshots will override your local publishing. It would also be helpful to highlight SBT commands to help see what versions of things are being used.

No worries. I'm happy to create a PR that updates the README with the full build instructions.

closed 2020-07-29 by mergify[bot]