chipsalliance/chisel #1552: Typo in the wiki of Bundles and Vecs

hadirkhan10, 2020-08-13:
**Type of issue**:  documentation

**Impact**: no functional change

**Development Phase**: proposal

**What is the use case for changing the behavior?** Better understanding of the documentation

hadirkhan10, 2020-08-13:
Here is the fixed typo mistake for the wiki. 
https://github.com/hadirkhan10/chisel3-wiki

seldridge, 2020-08-13:
Thanks for finding this and showing the correction @hadirkhan10! 

I manually applied your fix here: https://github.com/freechipsproject/chisel3/wiki/Bundles-and-Vecs/_compare/4977c335a2bba9dbec038b43d316aa330391ace3

As a note: the wiki is likely outdated and not well maintained. Some time back, the wiki content was migrated to the Chisel website so that the examples could be compiled (so we know they aren't broken): https://www.chisel-lang.org/chisel3/. The website *does* have your fix (via https://github.com/freechipsproject/www.chisel-lang.org/pull/42), so things are good to go on that side.

closed 2020-08-13 by seldridge

hadirkhan10, 2020-08-13:
Alright thanks for pointing out @seldridge