chipsalliance/chisel #1564: chisel emitts expensive verilog

sajjadahmed677, 2020-08-23:
Hi everyone! i need to know that is there any efficient way to generate pretty readable verilog from chisel, in current version it generates a lot of temporaries and randomization.

edwardcwang, 2020-08-24:
Hi Sajjad, were you interested in readable or high-performant Verilog? While those sometimes overlap, we've (sometimes surprisingly) found that in many cases the Chisel/FIRRTL-optimized Verilog is actually optimized better by downstream tools compared to hand-written Verilog. Was there a specific circuit or structure that was causing issues?

lsteveol, 2020-09-21:
At the CHIPS workshop update last week (9/17/2020) Jack Koenig updated that Chisel 3.4 would have many naming improvements which should make for more readable code. I believe they are also working on support to remove the randomization `ifdef chains for registers (through a config variable), which should also help with readability.