chipsalliance/chisel #1565: Chisel tester with newest verilator as backend got a compile error

name1e5s, 2020-08-24:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Other information**
 
In [this commit](https://github.com/verilator/verilator/commit/fac89c5d6233f900d0e382bde21526400427e388) of verilator, the function `Verilated::flushCall` has split into `Verilated::runFlushCallbacks` and `Verilated::runExitCallbacks`. This breaking change leads to a compile error when we want to run PeekPokeTester against newest verilator on macOS.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Use `brew update && brew upgrade` to update your enviroment, then run `sbt test` in [chisel-template](https://github.com/freechipsproject/chisel-template), then you will see  the following text:
```
cd /Users/name1e5s/Documents/Code/chisel-template/test_run_dir/make_no_vcd && verilator --cc GCD.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY -O1 --top-module GCD +define+TOP_TYPE=VGCD +define+PRINTF_COND=!GCD.reset +define+STOP_COND=!GCD.reset -CFLAGS "-Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h" -Mdir /Users/name1e5s/Documents/Code/chisel-template/test_run_dir/make_no_vcd --exe /Users/name1e5s/Documents/Code/chisel-template/test_run_dir/make_no_vcd/GCD-harness.cpp
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.040/share/verilator/include -I/usr/local/Cellar/verilator/4.040/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h  -Os -c -o GCD-harness.o /Users/name1e5s/Documents/Code/chisel-template/test_run_dir/make_no_vcd/GCD-harness.cpp
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.040/share/verilator/include -I/usr/local/Cellar/verilator/4.040/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h  -Os -c -o verilated.o /usr/local/Cellar/verilator/4.040/share/verilator/include/verilated.cpp
/usr/bin/perl /usr/local/Cellar/verilator/4.040/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VGCD.cpp VGCD__Slow.cpp VGCD__Syms.cpp > VGCD__ALL.cpp
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.040/share/verilator/include -I/usr/local/Cellar/verilator/4.040/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h  -Os -c -o VGCD__ALL.o VGCD__ALL.cpp
ar -cr VGCD__ALL.a VGCD__ALL.o
ranlib VGCD__ALL.a
/Users/name1e5s/Documents/Code/chisel-template/test_run_dir/make_no_vcd/GCD-harness.cpp:101:14: error: no member named 'flushCall' in 'Verilated'
  Verilated::flushCall();
  ~~~~~~~~~~~^
1 error generated.
make: *** [GCD-harness.o] Error 1
make: *** Waiting for unfinished jobs....
[info] GCDTester:
[info] GCD
[info] - should calculate proper greatest common denominator (with firrtl)
[info] GCD
[info] - should calculate proper greatest common denominator (with verilator) *** FAILED ***
[info]   java.lang.AssertionError: assertion failed:
[info]   at scala.Predef$.assert(Predef.scala:223)
[info]   at chisel3.assert$.apply(Assert.scala:76)
[info]   at chisel3.iotesters.setupVerilatorBackend$.apply(VerilatorBackend.scala:275)
[info]   at chisel3.iotesters.Driver$.$anonfun$execute$2(Driver.scala:55)
[info]   at scala.runtime.java8.JFunction0$mcZ$sp.apply(JFunction0$mcZ$sp.java:23)
[info]   at logger.Logger$.$anonfun$makeScope$2(Logger.scala:168)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[info]   at logger.Logger$.makeScope(Logger.scala:166)
[info]   at logger.Logger$.makeScope(Logger.scala:127)
[info]   at chisel3.iotesters.Driver$.$anonfun$execute$1(Driver.scala:38)
[info]   ...
```

**What is the current behavior?**
Running `sbt test` will get a depressing compile error.

**What is the expected behavior?**
The test should runs correctly.

**Please tell us about your environment:**

- version: `3.3.1`
- OS: `64bit Mac OS X 10.15.6 19G2021`
- verilator: `4.040`

name1e5s, 2020-08-24:
If anyone found the same problem, you can firstly
```
brew uninstall verilator
```
and then
```
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/86447d2239a9ebc4d98607fd4ae5b034d3349c02/Formula/verilator.rb
```
to downgrade the verilator.

It is not a good solution, but it works.😂

danielkasza, 2020-11-13:
I just ran into this. Apparently, `flushCall` is deprecated and it got removed at some point, but it was added back for backwards compatibility. Unfortunately, Ubuntu 20.10 ships with the affected Verilator version.

This seems to fix it:

```diff
diff --git a/src/main/resources/chisel3/top.cpp b/src/main/resources/chisel3/top.cpp
index 4e9c1433..3e478511 100644
--- a/src/main/resources/chisel3/top.cpp
+++ b/src/main/resources/chisel3/top.cpp
@@ -8,7 +8,7 @@
 // Override Verilator definition so first $finish ends simulation
 // Note: VL_USER_FINISH needs to be defined when compiling Verilator code
 void vl_finish(const char* filename, int linenum, const char* hier) {
-  Verilated::flushCall();
+  Verilated::runFlushCallbacks();
   exit(0);
 }
```

...though 2 tests are still failing. I am also not sure when `runFlushCallbacks` was added.

LeiWang1999, 2021-04-09:
Any progress?
I got same issue with my macos, it seems that the version problem of verilator :

```zsh
Princeling-Mac at ~ ❯ verilator -version
Verilator 4.040 2020-08-15 rev v4.038-32-gbdecf6c4e
```

But verilator did work on my ubuntu18.04 with that, the version of verilator was 2017

name1e5s, 2021-04-09:
> Any progress?
> I got same issue with my macos, it seems that the version problem of verilator :
> 
> ```shell
> Princeling-Mac at ~ ❯ verilator -version
> Verilator 4.040 2020-08-15 rev v4.038-32-gbdecf6c4e
> ```
> 
> But verilator did work on my ubuntu18.04 with that, the version of verilator was 2017

@LeiWang1999 Upgrade verilator to v4.102 or later. They fixed this problem [here](https://github.com/verilator/verilator/commit/efbcb094ca6fd066fdc0d0008b434ef33f0c9e45).

sequencer, 2021-04-09:
related to verilator/verilator#2580
Fixed in v4.102

closed 2021-04-09 by sequencer

LeiWang1999, 2021-04-09:
@sequencer @name1e5s Thanks, solved.