chipsalliance/chisel #1569: [3.4.0-RC1] missing reference

ekiwi, 2020-08-25:
I have this example circuit: https://scastie.scala-lang.org/XSpo4Q6kRnO2LCO4mKtW4w

It elaborates successfully with Chisel 3.3.x but on 3.4.0-RC1 I get an internal error where Chisel is trying to get the value of a `_ref` which is `None`:

```
Caused by: java.util.NoSuchElementException: None.get
	at scala.None$.get(Option.scala:529)
	at scala.None$.get(Option.scala:527)
	at chisel3.internal.HasId.getRef(Builder.scala:228)
	at chisel3.internal.HasId.getRef$(Builder.scala:228)
	at chisel3.experimental.BaseModule.getRef(Module.scala:162)
	at chisel3.internal.HasId.recArg$1(Builder.scala:162)
	at chisel3.internal.HasId.$anonfun$computeName$5(Builder.scala:164)
	at scala.Option.flatMap(Option.scala:271)
	at chisel3.internal.HasId.buildAggName$1(Builder.scala:164)
	at chisel3.internal.HasId.$anonfun$computeName$6(Builder.scala:177)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at chisel3.internal.HasId.buildName$1(Builder.scala:174)
	at chisel3.internal.HasId.computeName(Builder.scala:188)
	at chisel3.internal.HasId.computeName$(Builder.scala:151)
	at chisel3.Data.computeName(Data.scala:275)
	at chisel3.RawModule.$anonfun$namePorts$1(RawModule.scala:45)
	at chisel3.RawModule.$anonfun$namePorts$1$adapted(RawModule.scala:44)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
	at chisel3.RawModule.namePorts(RawModule.scala:44)
	at chisel3.RawModule.generateComponent(RawModule.scala:68)
	at chisel3.Module$.do_apply(Module.scala:69)
	at chisel3.stage.ChiselGeneratorAnnotation.$anonfun$elaborate$1(ChiselAnnotations.scala:60)
	at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:625)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at chisel3.internal.Builder$.build(Builder.scala:622)
	at chisel3.stage.ChiselGeneratorAnnotation.elaborate(ChiselAnnotations.scala:60)
```

The circuit was working with an earlier `SNAPSHOT` release of `3.4` so this regression seems like it was recently introduced.

ekiwi, 2020-09-22:
This still seems to fail on `3.4-RC2`

chick, 2020-09-22:
RC2 is broken, wish I could delete it

Chick Markley
Staff Programmer -- Adept Lab
University of California, Berkeley
583-2 Soda Hall, Berkeley, CA 94720-1776
chick@berkeley.edu

jackkoenig, 2020-09-22:
I don't think this has anything to do with `RC2`, this is just an open bug that is not yet fixed.

ekiwi, 2020-09-22:
> I don't think this has anything to do with `RC2`, this is just an open bug that is not yet fixed.

Yes, I just wanted to call attention to it.

This seems to be a regression since this circuit works with Chisel 3.3

ekiwi, 2020-09-22:
So this seems to be related to the fact that I directly assign the `IO` to the submodule input.

If I first assign the `IO` to a `val` it seems to all work:

```diff
--- a/test/examples/VariableLatencyExamplesSpec.scala
+++ b/test/examples/VariableLatencyExamplesSpec.scala
@@ -147,13 +147,15 @@ class VariableLatencyToConst extends MultiIOModule with IsConstLatency {
   val lsb = Module(new RandomLatency)
   lsb.io.start := io.start
   lsb.io.dataIn := io.dataIn(31, 0)
-  lsb.randomDelay := IO(Input(chiselTypeOf(lsb.randomDelay))).suggestName("lsbRandomDelay")
+  val lsbRandomDelay = IO(Input(chiselTypeOf(lsb.randomDelay))).suggestName("lsbRandomDelay")
+  lsb.randomDelay := lsbRandomDelay
   val lsbBuffer = RegEnable(lsb.io.dataOut, lsb.io.done)
 
   val msb = Module(new RandomLatency)
   msb.io.start := io.start
   msb.io.dataIn := io.dataIn(63, 32)
-  msb.randomDelay := IO(Input(chiselTypeOf(msb.randomDelay))).suggestName("msbRandomDelay")
+  val msbRandomDelay = IO(Input(chiselTypeOf(msb.randomDelay))).suggestName("msbRandomDelay")
+  msb.randomDelay := msbRandomDelay
   val msbBuffer = RegEnable(msb.io.dataOut, msb.io.done)
 

```

jackkoenig, 2020-09-23:
Interesting, glad you were able to find a workaround. We need to fix this but it can probably wait until 3.4.1.

jackkoenig, 2021-02-09:
Randomly came across this, it has been fixed since 3.4.1 and remains fixed in 3.4.2.

See same example with 3.4.2 working: https://scastie.scala-lang.org/lhXp0aEsRsq2ho5F7Jw9vA

closed 2021-02-09 by jackkoenig