chipsalliance/chisel #1593: out of memory when converting circuits to strings

YangWang92, 2020-09-21:
Hi all,
I'm trying to generate a large chip with chipyard in chisel, and I got this error.
I have confirmed that the host server has enough DRAM, and the java heap set to 512G.

> [info] [1459.065] Done elaborating.
[error] (run-main-0) java.lang.OutOfMemoryError
[error] java.lang.OutOfMemoryError
[error]         at java.lang.AbstractStringBuilder.hugeCapacity(AbstractStringBuilder.java:161)
[error]         at java.lang.AbstractStringBuilder.newCapacity(AbstractStringBuilder.java:155)
[error]         at java.lang.AbstractStringBuilder.ensureCapacityInternal(AbstractStringBuilder.java:125)
[error]         at java.lang.AbstractStringBuilder.append(AbstractStringBuilder.java:448)
[error]         at java.lang.StringBuilder.append(StringBuilder.java:136)
[error]         at scala.collection.mutable.StringBuilder.$plus$plus$eq(StringBuilder.scala:183)
[error]         at chisel3.internal.firrtl.Emitter.$anonfun$new$2(Emitter.scala:191)
[error]         at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[error]         at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
[error]         at chisel3.internal.firrtl.Emitter.$anonfun$new$1(Emitter.scala:191)
[error]         at chisel3.internal.firrtl.Emitter.withIndent(Emitter.scala:186)
[error]         at chisel3.internal.firrtl.Emitter.<init>(Emitter.scala:191)
[error]         at chisel3.internal.firrtl.Emitter$.emit(Emitter.scala:9)

https://github.com/freechipsproject/chisel3/blob/c613c285c3e044370024a1b5cc0a0da1dbc3e5e6/src/main/scala/chisel3/internal/firrtl/Emitter.scala#L9

I guess it is related to the maximum size of a string in java. How can I work around this issue? Thanks a lot!

Ref:
https://stackoverflow.com/questions/52712321/outofmemoryerror-when-joining-a-list-of-strings-in-java

jackkoenig, 2020-09-21:
Can you provide instructions on how to reproduce this in ChipYard? It does indeed look like you're hitting the fundamental String size limitations on the JVM so this will take a bit of work to deal with properly.

YangWang92, 2020-09-22:
Hi Jack,
I'm working on a manycore project, and you can reproduce our codes likes this on chipyard 1.3.
I created 128 rocket chips in the same chiptop. Maybe it is a little bit weird to generate a "manycore" chip like this. Is there any better way to do this? (https://groups.google.com/g/chipyard/c/8kh5-jUzZL4)

Thanks! 
Yang

```scala
abstract class BaseChipTop()(implicit val p: Parameters) extends RawModule with HasTestHarnessFunctions {
  // ..............................
  val systemClock = Wire(Input(Clock()))
  val systemReset = Wire(Input(Reset()))
  // overwrite parameters
  var id = 0;
  for(id <- 1 to 128){
    val lSystem = p(BuildSystem)(p.alterPartial({
      case AsyncClockGroupsKey => p(AsyncClockGroupsKey).copy
    }))
    val system = withClockAndReset(systemClock, systemReset) { Module(lSystem.module) }
    withClockAndReset(systemClock, systemReset) {
      val (_ports, _iocells, _harnessFunctions) = p(IOBinders).values.flatMap(f => f(lSystem) ++ f(system)).unzip3
      // We ignore _ports for now...
      iocells ++= _iocells.flatten
      harnessFunctions ++= _harnessFunctions.flatten
    }
  }
}
```

edwardcwang, 2020-09-22:
If all 128 cores are exactly identical, you might consider using `CloneModuleAsRecord` to just instantiate it 128 times instead of generating 128 different modules.

YangWang92, 2020-09-22:
Hi @edwardcwang,
Thanks for your advices and I'm tring to instantiate cores with `CloneModuleAsRecord`.

jackkoenig, 2020-09-22:
@YangWang92 that will help, but also should check your network. Rocket Chip designs default to using crossbars which is probably not going to work with 128 cores.

YangWang92, 2020-09-28:
Hi @jackbackrack  Thanks for your suggestion! We build a self-defined interface to connect cores.

And we reduce the simulation scale to work around the issue. Thanks a lot!

closed 2020-09-28 by YangWang92