chipsalliance/chisel #1598: Reassigning implicit clock and reset for DFT purposes

lsteveol, 2020-09-22:
**Type of issue**: feature request | other enhancement

**Impact**: unknown

**Development Phase**: request 

**Other information**
 
Currently there seems to be no way to reassign the implicit ``clock`` and ``reset`` for ``Module``. I believe there is a valid use case for doing so, particularly with regards to DFT SCAN where the clock and reset would need to be controlled in scan mode. For large single clock designs this can likely be alleviated by a wrapper level that does this. For multi-clock domain and mixed signal designs where clocks are potentially being generated at various levels in the hierarchy it may be advantageous to have a way to reassign the clocks without the need for a wrapper or ``MultiIOModule/RawModule``.

When trying to reassign the clock with the following

```scala
clock := Mux(io.in2, clock, io.in.asClock)
```
The following error is reported
```
chisel3.internal.ChiselException: Connection between sink (Clock(IO in unelaborated SSVPll)) and source (Clock(OpResult in SSVPll)) failed @: Sink is unwriteable by current module.
```

In general the circuit I would like to implement for the ``clock`` and ``reset`` would be the following:
[image]

I don't expect Chisel to place/implement the above circuit, but mainly to allow it's creation.

I would likely make a class/companion object that handles this with the appropriate connections like the following:
```scala
(clock, reset) := MyDFTScanLogic(clock, reset, io.scan.mode, io.scan.reset_ctrl)
```

Does this seem like a reasonable request? Are there other alternatives that I have not thought about?

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Not currently a bug. Workarounds can exists.

**What is the current behavior?**
No support for redeclaring/reassigning the clocks.

**What is the expected behavior?**

**Please tell us about your environment:**

Chisel 3.4.0-RC2

**What is the use case for changing the behavior?**
Support for DFT clock muxing and reset control

edwardcwang, 2020-09-22:
How does something like this look for your usecase?

```
val myClock = Mux(io.in2, clock, io.in.asClock)
withClock(myClock) {
  val myFancyRegister = RegInit(0.U(16.W)) // <-- clocked off myClock and reset via implicit reset
  // logic logic logic
}
```

lsteveol, 2020-09-22:
Something like this would be a valid solution. It would require me to explicitly declare this logic each time (which is ok). I was thinking about having an extended ``Module`` that when a DFT Bundle (that I create) is seen it would implicitly perform the scan insertion on the clock/reset. It would be something like

```scala
MyModule extends Module {
  if hasScanBundle {
    clock := stuff
    reset := stuff
  }
}
```

This way I don't have to keep up with it. But DFT is only at certain levels so maybe this is ok.

edwardcwang, 2020-09-22:
Hmm, maybe something like this?

```scala
class MyRegularModule extends Module { ... }

class MyTop extends RawModule {
  // params, ios, ...
  val modclock = if (hasScan) { <dft scan clock> } else clock
  val myModule = withClock(modclock) { Module(new MyRegularModule) }
}
```

lsteveol, 2020-09-22:
The only issue with the above is I need the hardware to physically implement a mux in the case of scan insertion. My "hasScan" check was really to see if there was a Bundle that implements scan. Let me try a few things out to see what seems to work the best and I can update. The first suggestion with withClock[andReset] is probably what I'll end up doing.

lsteveol, 2021-03-12:
I'm closing this issue, but wanted to show my solution. It is basically what @edwardcwang proposed. In my case we will actually hand place clock cells based on the process node, so I do the following

```scala
class ClockMux [T <: Data](a : T, b : T)(implicit p: Parameters) extends Module{
  //Compute the larger of the width and set the parameter
  val outType = if(a.getWidth > b.getWidth) a.cloneType else b.cloneType
  
  val io = IO(new Bundle{
    val i_sel   = Input (Bool())
    val i_a     = Input (a.cloneType)
    val i_b     = Input (b.cloneType)
    val o_z     = Output(outType.cloneType)
  })
  
  if(p(StdcellBehavioral)) {
    io.o_z := Mux(io.i_sel, io.i_b, io.i_a)
  } else {
    val wmux = Module(new clk_mux(outType.getWidth))

    wmux.io.i_sel   := io.i_sel
    wmux.io.i_a     := io.i_a
    wmux.io.i_b     := io.i_b
    io.o_z          := wmux.io.o_z
  }
  
}

```

In this case I have  a key that dictates if we instantiate our specific clock mux verilog wrapper or I can just use a behavioral.

closed 2021-03-12 by lsteveol