chipsalliance/chisel #1622: Cat of 0 element Vec throws exception on empty.tail

hcook, 2020-10-15:
**Type of issue**: bug report 

**Impact**: no functional change 

**Development Phase**: request 

**Other information**

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
https://scastie.scala-lang.org/ez6cIEduSkyz9hoct60urg

**What is the current behavior?**

Using the `Cat` operator on a `Vec` with zero elements throws an exception `java.lang.UnsupportedOperationException: empty.tail`

**What is the expected behavior?**

I would expect to get back an equivalent zero element `Vec`.

**Please tell us about your environment:**
https://scastie.scala-lang.org/ez6cIEduSkyz9hoct60urg

**What is the use case for changing the behavior?**
I don't want to have to write if statements guarding all my cats.

seldridge, 2020-10-15:
@hcook: Do you expect this to be a zero-element `Vec` or a `0.U`? I think you actually mean the latter here?

Thanks for the test case. That module name needs to be preserved for prosperity.

hcook, 2020-10-15:
Ahaha I was copying one of @jackkoenig's scastie examples and just left it. The original was pretty impressive tbf.

You're right that `0.U` is the most sensible output.

PR #1623 (2020-10-19): Enable Cat of Zero Element Vec
Prevent an exception thrown if you apply `Cat` to a zero-element `Vec`. This will now return `0.U` if you do that.

Fixes #1622 

### Contributor Checklist

- [x] Did you add Scaladoc to every public function/method?
- [x] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you specify the type of improvement?
- [x] Did you add appropriate documentation in `docs/src`?
- [x] Did you state the API impact?
- [x] Did you specify the code generation impact?
- [x] Did you request a desired merge str [... cut]
closed 2020-10-19 by mergify[bot]